nodemcu/nodemcu-firmware #1399: The existing SPIFFS code uses 4 as the number of sysparam sectors in a couple of places

pjsg, 2016-07-14:
Just a tracking issue to tell me to fix the code. With 1.5.4.1 the number of system parameter sectors is increasing to 5.

In one place, the number is 16384 (which is only used when we are operating in the 1M OTA upgrade mode. This should be changed (I think) to 5 * 4096.

The other case is in the spiffsimg tool where the expression is 4 * 4096.

marcelstoer, 2016-07-19:
Philip, will you be able to fix this within the next two weeks? I'm trying to figure out whether we should wait for SDK 1.5.4.1 compatibility before the next master snap or whether we should go with 1.5.4.

pjsg, 2016-07-19:
Yes -- there is a branch which contains the fix -- Probably best to merge it into the 1.5.4.1 branch. It is https://github.com/nodemcu/nodemcu-firmware/compare/dev...pjsg:correct-sector-count

Not easy to test standalone.....

marcelstoer, 2016-07-19:
I see, thanks. That'd be task for @devsaurus then, it's his branch.

devsaurus, 2016-07-19:
Thanks Philip, will pull it into my PR branch soon.

marcelstoer, 2016-07-20:
Arnim, have you had a chance to test this? If so, can I ask you to close it. I don't get around to verify (in combination with 1.5.4.1) myself today and tomorrow.

devsaurus, 2016-07-20:
It's in my PR branch for 1.5.4.1 and worked well with some smoke tests. Flash formatting and storing some files.

closed 2016-07-20 by devsaurus