nodemcu/nodemcu-firmware #1404: Use u8g module with SPI0

GelirDurum, 2016-07-18:
I working on nokia 5110 LCD screen with NodeMCU u8g module and my ESP version is ESP12-E. I tried my LCD by [u8g_graphics_test.lua](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/u8glib/u8g_graphics_test.lua). It' is worked perfectly. But I use this only HSPI (or SPI1 by use gpio 13 for MOSI and gpio14 for CLK). I tried this by SPI0 and changed pin connections for this (i use MOSI and CLK pin bottom side of board). After i changed codes only this line
 `spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)` 
  to 
`spi.setup(0, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)`. But this not worked.

 
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
Use SPI0 with u8g module
### Justification
This can add more useful gpio (gpio 13, gpio 14).

8<------------------------ END FEATURE REQUEST --------------------------------

devsaurus, 2016-07-18:
This would require overlap support in SPI driver as far as I understand.
[HSPI PORT OVERLAP MODE DEMO](http://bbs.espressif.com/viewtopic.php?t=56):
> In this mode , the two spi modules share the same pins

I'm not aware of any related activity for that.
Furthermore, the SS/CS handling in u8g would also need some update.

FrankX0, 2017-01-02:
This seems to be the same issue as #1602, not being able to use SPI0 because of a conflict with the attached Flash. Propose to close.

closed 2017-01-02 by marcelstoer