nodemcu/nodemcu-firmware #1405: GPIO 9 pin GND problem

GelirDurum, 2016-07-18:
8<------------------------ BUG REPORT -----------------------------------------
### Actual behavior
When i try write gpio9 to GND level esp is stopped a few second and after then restart itself. 
### Test code
```Lua
gpio.mode(11, gpio.OUTPUT)
gpio.write(11,0)
```
### NodeMCU version
branch: dev
built on: 2016-07-12 06:23
SDK: 1.5.4(baaeaebb)
### Hardware
Standart ESP8266 - 12F connections. Gpio 9 pin connect no where.

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-07-18:
Please confirm that your module is configured to [DIO mode](http://nodemcu.readthedocs.io/en/dev/en/hardware-faq/#whats-the-different-between-dio-and-qio-mode). If not then your commands interfere with the flash and this failure is not related to the firmware.

GelirDurum, 2016-07-19:
I'm not sure this but i think my module has configured for DIO mode. Because my Nodemcu firmware progremmer default spi mode is DIO  and i can control gpio 10 pin by same commands.

marcelstoer, 2016-07-19:
I can reproduce this using a [WeMos D1 mini](http://www.wemos.cc/Products/d1_mini.html#pin) and 1.5.4 but that's because pin 11 is not available on that board (firmware with DEBUG=ON):

```
gpio.mode(11, gpio.OUTPUT)
pin,mode,pullup= 11 1 0
Pin data at mode: 11 60000828, 9 3 0, fffffffe
Function platform_gpio_mode() is called. pin_mux:1610614824, func:3gpio.write(11,0)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 25988, room 16 
tail 4
chksum 0x5e
load 0x3ffe8000, len 2268, room 4 
tail 8
chksum 0xa1
load 0x3ffe88dc, len 8, room 0 
tail 8
chksum 0x34
csum 0x34
��{�l�l�mode : sta(5c:cf:7f:0b:62:bf)
add if0

_flash_used_end:4027aab3
fs.start:80000,max:37c000
mount res: 0, 0
Task task_lua started.

NodeMCU 1.5.4 build unspecified powered by Lua 5.1.4 on SDK 1.5.4(baaeaebb)
lua: cannot open init.lua
```

You'd see the same if you used a NodeMCU devkit 1.0.

devsaurus, 2016-07-20:
I see the same here with a NodeMCU devkit 1.0: pin 11 causes a hang and restart.

@GelirDurum It's probably caused by the ESP-12x modules that are soldered on your, Marcel's and mine boards. If you look at http://smarpl.com/content/esp8266-esp-201-module-freeing-gpio9-and-gpio10 and other tutorials on the net, there's one thing they have in common: In order to use the additional GPIOs, the flash must not be connected to these. Toggling of the GPIOs will crash the system in case they are connected (even if DIO mode is configured - my previous post was misleading, sorry).

I can't investigate the traces on my module because it's canned. But from the fact that my board works in QIO mode, I conclude that GPIO9 and GPIO10 **are** connected to the flash and thus there's no way to use them reliably.

closed 2016-07-20 by marcelstoer