nodemcu/nodemcu-firmware #1407: ethernet FCS error

bastos1505, 2016-07-20:
I used the wireshark program to sniff my esp8266 ESP12E and check why some devices couldn't connect into. The wireshark pointed that there is a problem in the ethernet FCS.
I'm using a version compiled in 2016/07/02.
My ESP is being used as SOFTAP;
I had seen in the source code and didn't find any code about the ethernet layer. I believe that it is inside a lib.

**### Expected behavior**
It was expected the value 0xD117F327 for FCS

**### Actual behavior**
The actual value was 0x00800006

**### Test code**
The wireshark log for ethernet is the following:
Destination: Broadcast (ff:ff:ff:ff:ff:ff)
Source: 64:89:9a:04:13;76
Type: Ipv4 (0x0800)
**_Frame check sequence: 0x00800006 [incorrect, should be 0xD117F327]_**

**### NodeMCU version**
NodeMCU custom build by frightanic.com
	branch: master
	commit: b580bfe79e6e73020c2bd7cd92a6afe01a8bc867
	SSL: false
	modules: cjson,file,gpio,http,net,node,sntp,tmr,uart,wifi
 build 	built on: 2016-07-02 17:29
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

### Hardware
I'm using ESP8266 ESP-12E.
I didn't find any problem with VCC or any kind of noise.

pjsg, 2016-07-23:
What was your exact setup? I wouldn't trust all adapters to report the actual FCS to wireshark...

devsaurus, 2016-07-23:
I was able to reproduce the incorrect FCS message from Wireshark. It's my Linux box contacting the ESP from a WIFI USB dongle. Ref attached [dhcp_fcs_error.pcap.zip](https://github.com/nodemcu/nodemcu-firmware/files/379492/dhcp_fcs_error.pcap.zip)

I see this only for DHCP packets and it doesn't occur for telnet traffic. Issue is maybe specific to either DHCP server or UDP protocol implementation (but not TCP).
FCS interpretation in general seems a bit fuzzy - Wireshark will interpret trailing bytes as FCS, but it won't look/check for a FCS if there are no trailing bytes in the frame.

This is exactly the point where I'm stuck at. ESP's DHCP server intends to send a packet with 548 bytes length. If the frame ended after this payload, all would be fine regarding FCS. But it doesn't end - there are additional 16 bytes which stimulate Wireshark's FCS mechanism. I've got no clue at the moment why they are present and which component generates them.

bastos1505, 2016-07-23:
Perfect devsaurus, that's it exaclty that happens with me!

My setutp is:
ESP8266 as SOFTAP
My notebook-LINUX OS-  as CLIENT (sniffing)
My cellphone as CLIENT-Android- (this one is able to connect)
My wife cellphone as CLIENT-Android-(this one don't connect)

Thank you guys.

djphoenix, 2017-01-19:
In latest SDK (`2.0.0` for now), and in `1.5.4.1`, was some fixes in LWIP, and in dhcp.

@bastos1505 @devsaurus can I ask you to re-check this issue on latest `dev`? No issues for my setup (macbook air with wireshark, and so iOS-clients works OK).

bastos1505, 2017-01-20:
For sure Bro. I'll check it this weekend.
BR,Carlos

closed 2017-11-03 by bastos1505