nodemcu/nodemcu-firmware #1409: Upgrade to SDK 2.x?

marcelstoer, 2016-07-20:
Espressif release NONOS SDK 2.0 today: http://bbs.espressif.com/viewtopic.php?f=46&t=2451

In light of the work @jmattsson et.al. started in #1319 does it even make sense to jump on the 2.x bandwagon?

jmattsson, 2016-07-21:
I think it will depend on whether the RTOS SDK gets some love too; I asked the question on Twitter, we'll see if there's a response.

jmattsson, 2016-07-21:
Oh, and it seems now everyone including Espressif are agreeing that mbedTLS is the way to go these days.

I didn't see it in the release notes, but it featured prominently in the [twitter announcement](https://twitter.com/EspressifSystem/status/755794324030271489).

djphoenix, 2016-07-21:
Will try it today at late evening (GMT+3). It may be helpful to #1379.

djphoenix, 2016-07-21:
> Oh, and it seems now everyone including Espressif are agreeing that mbedTLS is the way to go these days.

So, I found refs to libmbedtls in LD script, but no one included in SDK right now, in `lib`s nor `examples`.

djphoenix, 2016-08-02:
Checked this for some time. All seems to work. Propose to update, will make PR tomorrow.

marcelstoer, 2016-08-22:
Discussion can be continued in #1435 -> closing.

PR #1435 (2016-12-11): Upgrade to SDK 2.0.0
Fixes #1409, #1415, #1624, #1604
closed 2016-08-22 by marcelstoer