nodemcu/nodemcu-firmware #1411: How to keep up with we branched from?

marcelstoer, 2016-07-21:
When I looked into #1410 I wondered if by any chance it might have been "fixed" in the [original HTTP client](https://github.com/Caerbannog/esphttpclient) our code stems from.

So I compared [our httpclient.c](https://raw.githubusercontent.com/nodemcu/nodemcu-firmware/dev/app/http/httpclient.c) with [their httpclient.c](https://raw.githubusercontent.com/Caerbannog/esphttpclient/master/httpclient.c) and visualized it at http://nodemcu-build.com/httpclient.c.diff.html.

It's not fixed their either but I wondered on a more general level how we can/should keep up with we branched from. It's usually not as easy as cherry-picking from some other repo/branch. The longer the two versions of the code live an independent life the less likely it is that somehow automatically applying a patch-like artifact will work.

The real challenge though is learning about new fixes in the pseudo-upstream repo in the first place. Even in this particular case it's unlikely the original author(s) will just give up their own repo and keep using ours.

pjsg, 2016-07-23:
For the spiffs code, I ended up with a script (that is now part of the repo) that would import the latest version from github into the nodemcu repo. The files are then committed without any changes. There is a bit of preprocessor magic to make it compile for nodemcu. This makes it maintainable...

djphoenix, 2016-07-23:
[esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) using git submodules with custom makefiles and include overrides for this purpose. For example `mbedtls` library. Maybe take a look to similar solution.

marcelstoer, 2016-07-23:
We talked about Git submodules in #799 but for a slightly different purpose. The discussion here is about code that we _branch off_ from i.e. we have our own copy. The main purpose of somehow keeping an eye on the code origin is to learn about bug fixes there before we have to fix the same stuff ourselves in our own copy.

djphoenix, 2016-07-23:
We may use git submodules with **forked** modules, with apply of our own patches. In this case we still can fix and patch anything, but keep ability to fast merge changes from upstream.

closed 2016-11-19 by marcelstoer