nodemcu/nodemcu-firmware #1415: Update of LWIP code

djphoenix, 2016-07-23:
I took a look on `examples/lwip_open_src_template_proj` bundled in 1.5.4 SDK.
When copying it in place of nodemcu LWIP directories some changes appeared.
Most of them are not really changes something (e.g. tabs/spaces or endlines mismatch), but there are some interesting places...
Maybe we need to merge it.
Anyway there is [updated branch](https://github.com/djphoenix/nodemcu-firmware/tree/lwip-sdk-154). Please take a look on and review it. I will strip out unnecessary changes in next time.

djphoenix, 2016-07-23:
Cleaned up and forced update.

djphoenix, 2016-08-02:
@marcelstoer @devsaurus any reviews/comments? Should I make a PR or you'll write some script for autoupdate?
Anyway, Why we're using this option instead of bundled liblwip.a?

devsaurus, 2016-08-03:
>  Anyway, Why we're using this option instead of bundled liblwip.a?

There've been quite some discussions on the issue and PR lists regarding this topic. I've never dug into the details, but @jmattsson and @TerryE were the main drivers IIRC.

jmattsson, 2016-08-04:
Because at least with the source version we can fix bugs. It's a serious pain to carry said fixes forward across version upgrades, but it has so far been seen as the lesser evil.

TerryE, 2016-08-04:
+1 to this.  Sorry guys but I've been taking an enforced sabbatical from the project because I am working 12 hrs a day 7 days a week on my self-build new house at the moment :(  I would be far more comfortable using the open-source version, but we do need to track updates.

djphoenix, 2016-08-08:
Covered by #1435.

closed 2016-08-08 by djphoenix