nodemcu/nodemcu-firmware #1416: Feature request: software timers

djphoenix, 2016-07-23:
### Missing feature
Now we have 7 (0-6) hardware timer control functions. HW timers are pretty, but when we need more, we stuck in refactoring.

### API proposal
```lua
local timer = tmr.soft( function(t) -- self in callback
  print("TICK")
end )
timer:arm(1000, true) -- delay, [repeat=false]
timer:arm_us(152342) -- delay_us, [repeat=false]
timer:disarm()
timer = nil -- deallocated if not armed, kept otherwise
```

### Implementation proposal
I have finished it in my local repo (on top of `ETSTimer` of course), if there is no rejections of idea, I'll cleanup code, write docs and make PR.

devsaurus, 2016-07-24:
@djphoenix Similar ideas were discussed in #709 but there was never a conclusion on how to continue. Personally, I clearly favour an object-oriented solution like yours. Though with the current API in mind, how about the following approach for backwards compatiblity.

As you sketched above, `tmr.create()` creates a new timer object. It implements the existing tmr methods `alarm()`, `interval()`, `register()`, `start()`, `state()`, `stop()`, `unregister()` to keep its use model aligned with the existing timers. Non-timer functions are out of scope: `tmr.delay()`, `tmr.now()`, `tmr.softwd()`, `tmr.softwd()`, `tmr.time()`, `tmr.wdclr()`.

Much of the existing C code for the functions can be re-used / shared if they are augmented to accept a `self` userdata object in parallel to the numeric `id`. This allows to call them in any of these variants:
```lua
local mytimer = tmr.create()

-- oo calling
mytimer:register(5000, tmr.ALARM_SINGLE, function (t) print("expired"); t:unregister() end)
mytimer:start()

-- with self parameter
tmr.register(mytimer, 5000, tmr.ALARM_SINGLE, function (t) print("expired"); tmr.unregister(t) end)
tmr.start(mytimer)

-- still using an old timer
tmr.register(1, 5000, tmr.ALARM_SINGLE, function (t) print("expired"); tmr.unregister(t) end)
tmr.start(1)
```

Does this sound meaningful?

djphoenix, 2016-07-24:
Yep, this is good compromise. Let's try it in real world.

marcelstoer, 2016-07-24:
👍 for backwards compatibility. However, I'd like to see the old style functions tagged with a deprecation note both in code and the docs. That'll allow us to remove them maybe half a year later.

PR #1424 (2016-08-02): Dynamic timers support
Fixes #1416.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

I'll write docs soon, now will be great if anyone test this stuff.
closed 2016-08-02 by marcelstoer