nodemcu/nodemcu-firmware #1417: [FR/Q] Adding external C modules into NodeMCU env

blelump, 2016-07-24:
### Missing feature/documentation

Hi!

I wonder what's the best way of adding external C modules into NodeMCU environment. 

### Justification

I think I am not the first one who needed to extend the NodeMCU libraries stack with something that is not a part of the firmware, but I haven't found any suitable solution for such approach. 

### Workarounds

I've compiled the firmware with added external module. In particular it was [lua-stringy][1] (needed by [Multipart][2]) and what I've ended up with is:
```
static const LUA_REG_TYPE stringy[] = {
    {LSTRKEY("count"), LFUNCVAL(count) },
    {LSTRKEY("find"), LFUNCVAL(find) },
    {LSTRKEY("split"), LFUNCVAL(split) },
    {LSTRKEY("strip"), LFUNCVAL(strip) },
    {LSTRKEY("startswith"), LFUNCVAL(startswith) },
    {LSTRKEY("endswith"), LFUNCVAL(endswith) },
    { LNILKEY, LNILVAL }
};

NODEMCU_MODULE(STRINGY, "stringy", stringy, NULL);
```

So I was able to _inject_ `stringy` into the environment. Not sure this is the best way though. 

[1]: https://github.com/brentp/lua-stringy
[2]: https://github.com/Mashape/lua-multipart

devsaurus, 2016-07-24:
Your approach is state-of-the-art.

BTW, Marcel started a wiki page to show the details [How to write a C module](https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module). Guess it'll be included into the docs at some point.
Feel free to update the wiki page with your learnings. It's a better place to collect the know-how than in an issue.

marcelstoer, 2016-07-24:
I wish I could take credit for the wiki page 😉 It was provided by @romanchyla, though.

blelump, 2016-07-24:
@marcelstoer , @devsaurus , thanks guys !

marcelstoer, 2016-07-24:
Oh and once you're done with the multipart module (incl. the string functions) feel free to send us a PR 😉

blelump, 2016-07-24:
@marcelstoer , well , actually  Im still hoping that Ill find a solution out of the box. Right now Im doing a research with existing libs but no luck yet.