nodemcu/nodemcu-firmware #1418: understanding Question: sntp, runs sync continuously

aeickho, 2016-07-24:
```
NodeMCU 1.5.2+alex build unspecified powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
+rtctime
+sntp
```
I can't find the Information:
- If it necessary to call sntp.sync() periodically (to compensate the internel clock drift)  or is the epochcounter on sync automatically.

marcelstoer, 2016-07-24:
This is actually one for http://nodemcu.readthedocs.io/en/master/en/support/ - unless you'd want us to make the docs more explicit 😉 

> If it's necessary to call sntp.sync() periodically

Based on the nice [rtctime docs](https://nodemcu.readthedocs.io/en/master/en/modules/rtctime/) the answer is probably "yes" but I'd have to rely on @jmattsson for confirmation.

devsaurus, 2016-07-24:
The [documentation for `sntp.sync()`](http://nodemcu.readthedocs.io/en/master/en/modules/sntp/#sntpsync) is probably not precise enough - I had to look it up from time to time myself.
There's no background task or whatsoever to perform periodic synchronization. Sync between NTP server and the internal RTC happens only upon invocation of `sntp.sync()` (if `rtctime` is compiled in).

jmattsson, 2016-07-25:
What @devsaurus said ^^^

marcelstoer, 2016-07-25:
I'll add a note to the docs later today.

marcelstoer, 2016-07-25:
It's in the dev docs now: http://nodemcu.readthedocs.io/en/latest/en/modules/sntp/

closed 2016-07-25 by marcelstoer