nodemcu/nodemcu-firmware #1419: Truncated request content for multipart/form-data

blelump, 2016-07-25:
### Expected behavior

Hi!

When sending request with content type given as `multipart/form-data`, it seems the request body is truncated. If you have an idea where to look at it, give me a hint, because I haven't touched C for years.

For such request:
```
curl -F "text=default" <node address>
```
I'd expect the result
```
POST / HTTP/1.1
User-Agent: curl/7.35.0
Host: localhost:56500
Accept: */*
Content-Length: 146
Expect: 100-continue
Content-Type: multipart/form-data; boundary=------------------------4689fb342e685f00

--------------------------4689fb342e685f00
Content-Disposition: form-data; name="text"

default
--------------------------4689fb342e685f00--
```

### Actual behavior

The actual result is
```
POST / HTTP/1.1
User-Agent: curl/7.35.0
Host: 192.168.43.180
Accept: */*
Content-Length: 306
Expect: 100-continue
Content-Type: multipart/form-data; boundary=------------------------fa92aeba5cf935bd

```

### Test code
```
    sv = net.createServer(net.TCP, 30)
    sv:listen(80, function(c)
        c:on("receive", function(c, payload)
            print(payload)
            local response = "HTTP/1.0 200 OK\r\nServer: NodeMCU on ESP8266\r\nContent-Type: text/html\r\n\r\n OK"
            c:send(response)
            c:on("sent",function(conn)
                conn:close() 
            end)
            response = nil
        end)
    end)
```
and send whatever request you'd like with appropriate content type `Content-Type: multipart/form-data`; for instance:
```
curl -F "text=default" <node address>
```

You could examine the actual (desired) request with `nc`, eg:
```
nc -p 8282 -l
# and then
curl -F "text=default" <node address:8282>
```

### NodeMCU version
Custom build with Docker img. 
NodeMCU 1.5.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

### Hardware
ESP8266 12E

marcelstoer, 2016-07-25:
A dup of #960?

blelump, 2016-07-25:
@marcelstoer , thanks for answering.

Perhaps, I don't know for sure. Any ideas how to debug it ?

blelump, 2016-08-05:
@marcelstoer , any news on this? I can look at it, but I'd need some help where to start.

marcelstoer, 2016-08-05:
Sorry, I can't help you with this as I'm not familiar with the C code. @devsaurus, @jmattsson, @djphoenix can one of you help out?

djphoenix, 2016-08-05:
You are sending request and closing connection after receiving **first chunk** from socket. I suspect that http clients may send headers and payloads in multiple frames.

blelump, 2016-08-05:
@djphoenix , thanks for answering. Well, it's likely it's happening in such manner, however I'd be grateful for some hints where to start looking at. Any thoughts?

marcelstoer, 2016-08-05:
If I'm not mistaken it boils down to what @dvv reported in #960: the `receive` callback is expected to either fire more than once or only fire when the entire payload has been processed in the net layer. True, `multipart/form-data` (this issue) and `text/event-stream` (other issue) are two different content types but the fundamental issue is the same AFAICS.

marcelstoer, 2016-08-06:
I finally sat down and did some testing instead of just speculating and guessing. Turns out Yury was correct (but I had missed his relevant comment, sorry).

The `receive` callback would fire _twice_ (i.e. at every frame) if you let it. However, since you close the socket after the first frame was processed you don't get the second. Here's the output with DEBUG enabled:

```
net_server_connected is called.
net_on is called.
net_socket_received is called.
POST / HTTP/1.1
Host: 192.168.0.38
User-Agent: curl/7.43.0
Accept: */*
Content-Length: 146
Expect: 100-continue
Content-Type: multipart/form-data; boundary=------------------------b843b8317e8fa161

net_socket_received is called.
--------------------------b843b8317e8fa161
Content-Disposition: form-data; name="text"

default
--------------------------b843b8317e8fa161--

net_server_disconnected is called.
```

It was produced by the following small script:

```lua
srv = net.createServer(net.TCP)
srv:listen(80, function(conn)
    conn:on("receive", function(sck, payload)
        print(payload)
    end)
end)
``` 

See how "net_socket_received is called." is printed twice.

Conclusion: you need to track the frames manually and only close the socket once you've received all data (check `Content-Length`?).

closed 2016-08-06 by marcelstoer

blelump, 2016-08-07:
@marcelstoer , thanks for pointing this out!

#### Summary

The reason I've started with `multipart/form-data` is that I wanted to upload files remotely to the controller. Such encoding helps here, because I wanted to specify which file has to be overidden with the incoming data, hence I'd be able to upload any file remotely. However it seems it's not easy to achieve since it would be nice to have request body given as a string to decode the incoming data and extract file name and file content appropriately. This leads straight to the `string length overflow` issue if the file content is large enough. On the other hand, each incoming body chunk (of size 1460 bytes) can be stored in an array, however to properly decode the whole body, utilizing string operations is desired or at least 'nice to have'. I can imagine decoding body stored as an array, but this is way too much for my needs.

To sum it up, `multipart/form-data` seems to be a feature that is too big for the current environment, or at least that's my current understanding of the underyling problem. I cannot imagine any other use case for `multipart/form-data` except file uploading and hence I abandon further exploration.