nodemcu/nodemcu-firmware #1422: Help with extension of the crypto module

ameeuw, 2016-07-27:
Hello everybody, I need to have the secp256k1 signing algorithm on a esp8266 and would like to ask here for help to implement this.

As there are many implementations written in C I thought it would be easier to extend the crypto module instead of writing a lua module for the task.

Is anyone experienced in ECDSA algos and maybe familiar with the crypto module in NodeMCU and could help me find the right way to add it to the module?

Thank you.

marcelstoer, 2016-07-27:
In case you haven't found it yet http://nodemcu.readthedocs.io/en/latest/en/modules/crypto/ gives you a pointer to the source and the current maintainer. https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module has some general info as for how to create a C module. HTH.

marcelstoer, 2016-08-07:
@ameeuw Can we close this? Was the information I pointed you to enough to get you started?

jmattsson, 2016-08-11:
I'm familiar with the crypto module, but have never touched ECDSA. Tag me with any specific questions and I'll try to help out! :)  Currently very busy on other projects, so NodeMCU is not getting much time from me atm.

Do you have any links to the C implementation(s) you had in mind? It'd need to be something under a BSD-style license or public domain; we're not going down the GPL path at this point in time.

marcelstoer, 2016-08-22:
No feedback from author in nearly a month -> closing.

closed 2016-08-22 by marcelstoer