nodemcu/nodemcu-firmware #1425: Websocket module

luismfonseca, 2016-07-28:
### Missing feature
A native [Websocket ](https://tools.ietf.org/html/rfc6455) module.

### Justification
The websockets allow a two-way communication between the client and server, so it's more well suited for IoT applications where one would like to stream data.
Being an upgradable protocol from HTTP this means the existing infrastructure can still handle websockets connections.

The nodemcu offers an `net` module for more raw control of the connection - this a higher-level protocol that sits on the same level of abstraction of the `http`.

The main advantage for a native module is heap usage.

### Workarounds
There are 2 projects that implement this in Lua [[1]](https://github.com/creationix/nodemcu-webide) [[2]](https://github.com/ikaras/nodemcu-websocket-client). Both use more than 10KB of heap.

### Implementation
I'm aware this is a non-trivial effort, therefore I'm also volunteering to do this myself. Although I feel I should reach out for possible implementation discussion & comments.
I'm also more interested in a client version, although after a client is done, a server should not be too hard.

closed 2016-08-18 by marcelstoer

Governa, 2016-10-14:
I've found some problems with the websockets module while using secure connections. While it would work with something like wss://echo.websocket.org, it didn't with my tornado development server.

For some reason, its callbacks are called with an empty buffer and len == 0 and the code stops working. I've added some checks for those cases and it seems to work.

https://github.com/Governa/nodemcu-firmware/commit/c7f8c3bdf48f2c0a65a1d25afd39ae7078fde32f

marcelstoer, 2016-10-14:
@Governa Please raise a PR with your changes. Look at the SSL/TLS notes at http://nodemcu.readthedocs.io/en/latest/en/modules/net/ for a likely reason why secure connections might fail.

PR #1433 (2016-08-18): Feature Websocket module
Fixes #1425.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Websockets allow a two-way communication with servers which is better suited for realtime applications. Existing Lua impementations are not as fast and take over 10KB of heap for a single connection.

Since it has been a several years without programming in C or C++, I'm sur [... cut]