nodemcu/nodemcu-firmware #1427: Remove DEVKIT_VERSION_0_9

marcelstoer, 2016-07-31:
I want to remove the DEVKIT_VERSION_0_9 flag which only serves to enable/disable and configure the two deprecated functions `node.led()`, `node.key()`.

```
$ grep -r "DEVKIT_VERSION_0_9" *
app/include/user_config.h:// #define DEVKIT_VERSION_0_9 1 	// define this only if you use NodeMCU devkit v0.9
app/include/user_config.h:#ifdef DEVKIT_VERSION_0_9
app/modules/node.c:#ifdef DEVKIT_VERSION_0_9
app/modules/node.c:#ifdef DEVKIT_VERSION_0_9
docs/en/modules/node.md:This function is only available if the firmware was compiled with DEVKIT_VERSION_0_9 defined.
docs/en/modules/node.md:This function is only available if the firmware was compiled with DEVKIT_VERSION_0_9 defined.

```
Any objections? I never owned an old devkit. Hence, I never used either of the functions myself.

devsaurus, 2016-07-31:
What exactly do you intend - just remove the define from user_config.h or remove also the code for `node.led()` and `node.key()`?

marcelstoer, 2016-07-31:
Those functions have been deprecated for the longest time, more than a year. I don't even know whether they still work. So, IMO either we remove them or we remove the deprecation note.

devsaurus, 2016-07-31:
Ok, understood.
:+1: for finally removing the functions.

PR #1439 (2016-08-04): Remove DEVKIT_VERSION_0_9 and deprecated functions
Fixes #1427.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.
closed 2016-08-05 by marcelstoer