nodemcu/nodemcu-firmware #1428: exception reset since aa5fd1445d40d2d3f2846d1b876ab388492e57d0

eku, 2016-07-31:
I recently updated my nodemcu project to the HEAD of the dev branch. Unfortunately I got an immediately execption reset when executing Lua file with dofile('filename.lc'). I run git bisect to find the patch thathas broken my setup. Previous version has no problems.

### Expected behavior
no exception reset

### Actual behavior
exception reset since [aa5fd1445d40d2d3f2846d1b876ab388492e57d0] Change 'formatting file system' message (#1413) 

```
lua: cannot open init.lua
> =node.bootreason()
2       2       0       1076011944      0       0       1076219773      0
```

### Test code

This is a snippet from my Lua initialization script, which runs fine till aa5fd1445d40d2d3f2846d1b876ab388492e57d0.

```Lua
cfg={
 hostname="esp1",
 sta={
  mac="00:cb:00:66:df:ab",
  ip={
   ip="192.168.33.131",
   netmask="255.255.255.0",
   gateway="192.168.33.1"
  },
  ssid="STATION",
  pwd="1234567890123456"
 },
 ap={
  mac="1a:fe:43:2a:83:aa8",
  ip={
   ip="172.22.1.1",
   netmask="255.255.255.0",
   gateway="172.22.1.1"
  },
  ssid="ESP",
  pwd="password"
 },
 dns="192.168.22.1",
}

function setupSTA(ssid,pwd,mac,ipinfo)
 wifi.sta.setmac(mac)
 wifi.setmode(wifi.STATION)
 wifi.sta.config(ssid,pwd,1)
 wifi.sta.setip(ipinfo)
end

function setupAP(ssid,pwd,mac,ipinfo)
 wifi.ap.setmac(mac)
 wifi.setmode(wifi.STATIONAP)
 wifi.ap.config({ssid=ssid,pwd=pwd,auth=wifi.AUTH_WPA2_PSK})
 wifi.ap.setip(ipinfo)
end

setupSTA(cfg.sta.ssid,cfg.sta.pwd,cfg.sta.mac,cfg.sta.ip)
setupAP(cfg.ap.ssid,cfg.ap.pwd,cfg.ap.mac,cfg.ap.ip)

```
### NodeMCU version
NodeMCU 1.5.4.1 build 20160731 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

```
$ grep "^#define" src/app/include/user_modules.h
#define __USER_MODULES_H__
#define LUA_USE_BUILTIN_STRING          // for string.xxx()
#define LUA_USE_BUILTIN_TABLE           // for table.xxx()
#define LUA_USE_BUILTIN_COROUTINE       // for coroutine.xxx()
#define LUA_USE_BUILTIN_MATH            // for math.xxx(), partially work
#define LUA_USE_BUILTIN_DEBUG_MINIMAL // for debug.getregistry() and debug.traceback()
#define LUA_USE_MODULES_FILE
#define LUA_USE_MODULES_NET
#define LUA_USE_MODULES_NODE
#define LUA_USE_MODULES_PWM
#define LUA_USE_MODULES_TMR
#define LUA_USE_MODULES_UART
#define LUA_USE_MODULES_WIFI
```

### Hardware
ESP-12

devsaurus, 2016-07-31:
I get an error when running your code (integer build):

```
stdin:2: bad argument #1 to 'setmac' (MAC:FF:FF:FF:FF:FF:FF)
stack traceback:
	[C]: in function 'setmac'
	stdin:2: in function 'setupAP'
	stdin:1: in main chunk
> =cfg.ap.mac
1a:fe:43:2a:83:aa8
> 
```

Is the value for `cfg.ap.mac` valid at all?

Changing `aa8` to `a8` allows the code to execute without further complaints. Same for a float build.

eku, 2016-07-31:
@devsaurus you are right, my fault while _obfuscating_ the data for this issue. I still get an exception with the whole lua file. Will try to dig a bit deeper. Where do these execption addresses from ```node.bootreason()``` point to?

eku, 2016-07-31:
Put the folling code into ```test.lua```, compile it and run with ```dofile('test.lc')```

```
cfg={
 hostname="esp1",
 sta={
  mac="00:cb:00:66:df:ab",
  ip={
   ip="192.168.33.131",
   netmask="255.255.255.0",
   gateway="192.168.33.1"
  },
  ssid="STATION",
  pwd="1234567890123456"
 },
 ap={
  mac="1a:fe:43:2a:83:a8",
  ip={
   ip="172.22.1.1",
   netmask="255.255.255.0",
   gateway="172.22.1.1"
  },
  ssid="ESP",
  pwd="password"
 },
 dns="192.168.22.1",
}

--wlan=require('wlan')
--wlan.init()

function setupSTA(ssid,pwd,mac,ipinfo)
 wifi.sta.setmac(mac)
 wifi.setmode(wifi.STATION)
 wifi.sta.config(ssid,pwd,1)
 wifi.sta.setip(ipinfo)
end

setupSTA(cfg.sta.ssid,cfg.sta.pwd,cfg.sta.mac,cfg.sta.ip)
```

Gives me 

```
> dofile('t.lc')

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
```

Does not happen when typed into an interactove session. Could this be an timing issue?

devsaurus, 2016-07-31:
> Where do these execption addresses from node.bootreason() point to?

One of these should point to the instruction that caused the exception. Can't remember if it's [`epc1` or `excvaddr`](http://nodemcu.readthedocs.io/en/dev/en/modules/node/#returns), probably the latter.

If you compiled locally then `app/mapfile` allows you to narrow down the address to a specific function. More detailed localization can be obtained from the disassembler dump:

```
$ xtensa-lx106-elf-gdb app/.output/eagle/debug/image/eagle.app.v6.out 
(gdb) disass <function_name>
```

That's what I had in mind when trying to reproduce the crash. But I'm not yet there...

devsaurus, 2016-07-31:
To rule out some potential pitfalls:

Have you compiled from a clean tree? I had headaches in the past when updating the code base and skipped `make clean`. Even `node.heap()` behaved unreliable in that case.

Init data is [updated](http://nodemcu.readthedocs.io/en/dev/en/flash/#upgrading-firmware)? Though I don't think a lack of it will cause random crashes - firmware wouldn't start at all.

devsaurus, 2016-08-01:
@eku do you have any news on this? Your report questions fundamental functionality and I'd like to understand the issue before signing off the master drop at #1431.

eku, 2016-08-01:
@devsaurus still investigating.I am not practiced in troubleshooting on machine instruction level. Stay tuned...

eku, 2016-08-01:
@devsaurus erased the flash as stated in the [update section](http://nodemcu.readthedocs.io/en/dev/en/flash/#upgrading-firmwarel). Now the module is dead. No lua prompt anymore. Damn!
Cant find a file called ```esp_init_data_default.bin```in the repository. Any hint?

devsaurus, 2016-08-01:
Allow some time for erasing the flash.
Flashing init data should work as fallback. Take it from the 1.5.4.1 patch ZIP in `cache` folder.

eku, 2016-08-01:
Unfortunately the fallback did not work. Went back to 8866297c6d7cc35afd25b0f0f666ef90af040300 and flashed ```esp_init_data_default.bin``` from the SDK 1.5.4 and now the module is alive again. Don't know why the 1.5.4.1 breaks my module. Will try to get forward....

eku, 2016-08-01:
I'm now back on ```HEAD``` with the correct ```esp_init_data_default.bin```. and the exception reset is also back :-1: Will dive into disassembly...

devsaurus, 2016-08-01:
Still no luck reproducing this.
Please post the compiled image somewhere together with the commands you use for flashing.

marcelstoer, 2016-08-01:
I can't reproduce this either. With the 1.5.4.1 update I found it's absolutely essential you get the esptool.py write parameters right (`-fm dio -fs 32m -ff 40m` for 4MB devices).

eku, 2016-08-02:
@marcelstoer  thanks for the hint. That did the trick.

I'm wondering why this is importent? I used ```make flash``` for tha last 12 months for modules with 512kB and 4MB without problems till SDK ```1.5.4.1```. Should we remove the rule  ```flash``` from the makefile?

marcelstoer, 2016-08-02:
> Should we remove the rule flash from the makefile?

Absolutely, see #595.

closed 2016-08-02 by marcelstoer

dnc40085, 2016-08-02:
>>Should we remove the rule flash from the makefile?

>Absolutely, see #595.

I like having `make flash`, it makes it easier to flash from the IDE.
I say :thumbsdown: to the removal of the make target.

marcelstoer, 2016-08-02:
I understand but this issue once again proved that `make flash` w/o all the esptool.py `write_flash` options is prone to fail or cause trouble later. So, at best `make flash` gives you a false sense of "coding comfort".

PR #1442 (2016-08-06): removed default flash rule
Fixes #1428

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

When flashing the firmware by using the rule ```flash``` in the ```Makefile``` can lead to an unstable behavior as observed in #1428. 

In o [... cut]