nodemcu/nodemcu-firmware #1432: SSL verification problem on ESP8266 side

sergbod, 2016-08-02:
Please, anybody help me to resolve below problem. If I'm not right - correct me.
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
"Client Hello" message from ESP to HTTPS server in TLSv1.1 must send actual GMT Unix Time in field "Random". 
### Actual behavior
"Client Hello" message from ESP to HTTPS server in TLSv1.1 must contains GMT Unix Time in field "Random" but actually it contains "Jan 1, 1970 02:00:00.000000000" in all messages without changings. And it seems that due to above mentioned after near 30 minutes of seccessefull work (messages sents every 10 seconds) the ESP sends message about failure "Bad certificate". My SSL certificate has one year of validity (starting from Jul 28, 2016). Wireshark dump screenshoot is attached also.
[image]

HTTPS server IP: 192.168.0.136
ESP client IP: 192.168.0.178

### Test code

```Lua
sntp.sync('ntp.time.in.ua',
  function(sec,usec,server)
    print('sync', sec, usec, server)
  end,
  function()
   print('failed!')
  end
)
net.cert.verify(true)
```
### NodeMCU version
Current actual version from master branch

### Hardware
ESP8266 (ESP-07, 1Mb flash)

8<------------------------ END BUG REPORT -------------------------------------

djphoenix, 2016-08-02:
Reason is not in random field. You was enabled certificate validation, but not set up CA cert to validate path. Correct me if I was wrong.

sergbod, 2016-08-02:
The CA certificate was uploaded to ESP Flash in initial stage in init.lua by means of net.cert.virify([[ text of certificate]]). First 30 min it works fine, but after - it gives bad certificate alert.

djphoenix, 2016-08-02:
Hmm. Was you synchronized time with SNTP? Looks like RTC time is empty, so timestamp is zero-filled.

sergbod, 2016-08-02:
In initial stage I'm doing synchronization with NTP server by means of sntp.sync(), and it gives correct results. After verification by rtctime.get() it gives correct results, and internal clock is working fine.

sergbod, 2016-08-02:
But it still gives zeros in timestamp in random field of TLS protocol. It seems that something wrong in function net.cert.verify()

djphoenix, 2016-08-02:
Oh, espconn... I have no idea for this. The one thing I can propose is wait for #1379 finished, so I will begin to work with mbedtls.

pjsg, 2016-08-03:
I think that waiting for mbedtls is the right approach. The existing code in the SDK is somewhat of a mess.

devsaurus, 2016-08-03:
@sergbod `net.cert.verity()` forwards verification to `espconn_secure_ca_enable()` which is in Espressif's domain. The SDK API Guide doesn't mention any prerequisites regarding ToD/sntp setup. Thus my guess is that it's ignoring anything related to current time.
@djphoenix and @pjsg sketched the possibilities for nodemcu firmware. Alternatively, you could ask Espressif directly for their opinion in the [General Q&A forum](http://bbs.espressif.com/viewforum.php?f=65).

closed 2016-08-07 by marcelstoer