nodemcu/nodemcu-firmware #1434: Wrong {} behavior

limpkin, 2016-08-02:
### Expected behavior
{} operator to pack multiple return values into an array

### Actual behavior
several values set to nil instead of their correct values

### Test code
Provide a minimal and isolated test which will reproduce the problem.
```Lua
1) string containing 60 uint32_t:
dada = "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"
2) use struct to unpack them:
data_to_send = {struct.unpack("iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii", dada)}
3) print them individually:
for i=1,60 do print(data_to_send[i]) end
808464432
808464432
808464432
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
nil
808464432
808464432
...
4) use print for the unpack returns:
print(struct.unpack("iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii", dada))
808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	808464432	241
```
### NodeMCU version
Both dev and main versions have this problem, as of right now.

### Hardware
standard esp12-e flashed with nodemcu FW

limpkin, 2016-08-02:
Bug present when there are more than 3 values to be returned.

Erroneous behavior:
`
four_uints = "aaaaaaaaaaaaaaaa"
packed_data = {struct.unpack("iiii", four_uints)}
for i=1,4 do print(packed_data[i]) end
1633771873
1633771873
1633771873
nil
`

Showing that the bug actually happens:
`
four_uints = "aaaaaaaaaaaaaaaa"
print(struct.unpack("iiii", four_uints))
1633771873	1633771873	1633771873	1633771873	17
`

djphoenix, 2016-08-02:
Ack this, found same behavior. Will search for corruption place...
As a note - index and count of nils are not stable. I found case where fifth element of 14 was nilled, but others are Ok.

limpkin, 2016-08-31:
Hello there,

Any update on this issue? I unfortunately can't do any high speed transfer through wifi if it isn't correct.

Thanks!

devsaurus, 2016-08-31:
I'd guess that the returned [number of pushed elements](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/struct.c#L364) is wrong. Apparently this doesn't confuse `print()` but corrupts table construction.

pjsg, 2016-09-01:
The code in the struct module looks good -- I added the struct module, but the code was essentially unaltered.....

devsaurus, 2016-09-01:
Agreed, Philip. I experimented with the test case and got the impression that the issue is not specific to struct. Other functions should exhibit the same behavior and I intend to try this next.

devsaurus, 2016-09-01:
The following test case should show the same "{} issue" if it was generic, but it doesn't:

```lua
spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8, spi.FULLDUPLEX)
packed_data = {spi.send(1, 1, 2, 3, 4, 5)}
for i=1,6 do print(packed_data[i]) end
5
255
255
255
255
255
```

pjsg, 2016-09-02:
I was running a DEVELOP build, and I got

ASSERT@lapi.c(447): L->top < L->ci->top

jmattsson, 2016-09-02:
There seems to be a missing `luaL_checkstack()` before the final push in `b_unpack()`, but that's not what's causing this issue. Are we managing to corrupt the stack and/or heap somehow here? The Non-OS SDK doesn't seem to have a canary between the two, so silent weirdness is a definite possibility that way...

pjsg, 2016-09-02:
I only got one copy of the assert (and I don't think that there is any duplicate detection to suppress it).

pjsg, 2016-09-03:
Changing the checkstack call to fix the overflow does not affect the outcome -- the nils still appear.

jmattsson, 2016-09-03:
Yeah in this case the `checkstack()` does nothing - there's enough room on the stack already. It seems the actual issue is related to the table-creation and populating. I'm not all that familiar with the Lua VM internals though, that's more your and @TerryE's speciality. At least we have a small, easily reproducible test case to work with :)

pjsg, 2016-09-04:
There is something horrible going on. I tried struct.unpack("cccccccccc", dada) and it threw more asserts than I knew what to do with.... As far as I can tell, "cccccc" should unpack one character at a time......

pjsg, 2016-09-04:
```
dada = "000100020003000400050006000700000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
pat = "ccccccccccccccccccccc"
d = {struct.unpack(pat, dada)}
for k,v in pairs(d) do print("index:"..k.." value:"..v) end
print(struct.unpack(pat, dada))
```

This a number of "c" unpack characters that really starts to cause problems. Between 3 & 19, the table that gets built only has 3 elements. At 19, it gets four elements (1,2,3 & 19) 

Bizarrely, the code
```
function f()
dada = "000100020003000400050006000700000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
pat = "ccccccccccccccccccccc"
d = {struct.unpack(pat, dada)}
for k,v in pairs(d) do print("index:"..k.." value:"..v) end
print(struct.unpack(pat, dada))
end

f()
```

(just wrapping in a function) -- this behaves differently (and still badly).

```
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
index:1 value:0
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
index:2 value:0
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
ASSERT@lgc.c(181): !iscollectable(&h->array[i]) || (ttype(&h->array[i]) == (&h->array[i])->value.gc->gch.tt)
index:3 value:0
index:4 value:1
index:5 value:0
index:6 value:0
index:7 value:0
index:8 value:2
index:9 value:0
index:18 value:0
index:19 value:0
index:20 value:index:3 value:0
index:21 value:0
index:22 value:22
0	0	0	1	0	0	0	2	0	0	0	3	0	0	0	4	0	0	0	5	0	22
```

If you run `f() again a few times, then the value of index:20 changes to other strings.....

pjsg, 2016-09-04:
By comparing the OP_SETLIST implementation (in lvm.c) with lua5.2, it appears that the fix is just moving a line of code, I've tried it and it *seems* to work. I'll post a PR later on today......

jmattsson, 2016-09-05:
Fix has now been merged into dev.

@limpkin Thanks for finding and reporting this one! Sorry it tooks us a while to get on top of it.

closed 2016-09-05 by jmattsson

limpkin, 2016-09-05:
Hello Johny,

No worries! I had other projects to take care of in the mean time... thanks for your time to all of you!

limpkin, 2016-09-09:
Thanks again to everyone, it works perfectly and allows me to send high volumes of data with a few lines of code without setting up (once again) a complete build environment.

If anyone was to come accross this issue, these are the lines:

`
-- UDP Server Data receive
function on_udp_packet_receive(srv, payload)
	-- Here we receive a 60 bytes array of int32_t, little endian
	-- We simply unpack the values and send them as is to the FPGA
	-- The SPI driver is set to 24 bits send
	if string.len(payload) == 240 then
		-- complete buffer update, timed at 5.8ms
		data_to_send = {struct.unpack("<!4iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii", payload)}
		-- delete last byte which contains parsing index
		data_to_send[61] = nil
		spi.send(1, data_to_send)	
`

devsaurus, 2016-09-10:
Interesting application. Are there more details available online? Regarding Fpga type, bitstream (post)processing, udp data transfer...

Sounds like an elegant way to  configure these over the air.

limpkin, 2016-09-11:
Of course! Unfortunately I have started a write-up on my website as I'm still waiting for part of the non-electronics hardware. To quickly summarize, it's a custom clock made of 4 7-segment displays and 60 nixie tubes connected to the internet.
The FPGA used is a spartan3 with onboard flash and a custom SPI slave controller implemented. It also features a 60 channels 14-bit PWM controller.
The ESP8266 running NodeMCU is therefore in charge of sending the PWM values and update the 7-segment displays using SPI.

Here's the complete code in case you're interested:
```Lua
--nixie_clock.lua
-- PWM ADDRs for SEGMENTS:
--	   8
--	  --
-- 6 |	| 9
--	  --  5
-- 4 |	| 2
--	  --  . 7
--	  3
-- minus starting addr (2):
--	   6
--	  --
-- 4 |	| 7
--	  --  3
-- 2 |	| 0
--	  --  . 5
--	  1

-- Below: for each number (0->9) a boolean array specifying if we should set a non zero value into the pwm register, following the PCA9624 pwm register order
number_to_reg_val = {{1,1,1,0,1,0,1,1},{1,0,0,0,0,0,0,1},{0,1,1,1,0,0,1,1},{1,1,0,1,0,0,1,1},{1,0,0,1,1,0,0,1},{1,1,0,1,1,0,1,0},{1,1,1,1,1,0,1,0},{1,0,0,0,0,0,1,1},{1,1,1,1,1,0,1,1},{1,1,0,1,1,0,1,1}}
digit_addresses = {0x08, 0x09, 0x0a, 0x0b}
displayed_digits = {0, 0, 0, 0}
start_digit_reg_addr = 2
dot_shown = false
sda, scl = 1, 2
led_noe = 4
-- Below: variables related to time getting
timeZone = 1 -- GMT+1 hours
hour = 0
minute = 0
second = 0
day = 0
month = 0
year = 0

-- returns the hour, minute, second, day, month and year from the ESP8266 RTC seconds count (corrected to local time by tz)
-- credit to kieranc in esp8266.com
function getRTCtime(tz)

	-- returns 2 if a given year is a leap year, else returns 1
	local function isleapyear(y)
	   if ((y%4)==0) or (((y%100)==0) and ((y%400)==0)) then
			return 2 
	   else 
			return 1 
	   end
	end
   
	-- returns 366 is a given year is a leap year else returns 365
	local function daysperyear(y)
	   if isleapyear(y)==2 then 
			return 366 
	   else 
			return 365 
	   end
	end		   

	-- determines whether Daylight Saving Time is in effect for a given hour, day, month and year
	local function isDST(hour,day,month,year)

		-- returns the date for Nth day of the month.
		-- for example: nthDate(2016,3,0,2) will return the date of the 2nd Sunday in March 2016 (the 13th) when DST begins
		--				nthDate(2016,11,0,1) will return the date of the 1st Sunday in November 2016 (the 6th) when DST ends
		local function nthDate(year,month,DOW,week)
[... 169 lines cut ...]
		-- get the hour, minute, second, day, month and year from the ESP8266 RTC
		hour,minute,second,month,day,year = getRTCtime(timeZone)
		-- display time
		displayed_digits[1] = hour/10
		displayed_digits[2] = hour%10
		displayed_digits[3] = minute/10
		displayed_digits[4] = minute%10	
		display_number_array()
		-- flip dot
		if dot_shown == true then
			dot_shown = false
		else
			dot_shown = true
		end
		--if year ~= 0 then
		--	-- format and print the hour, minute second, month, day and year retrieved from the ESP8266 RTC
		--	print(string.format("%02d:%02d:%02d	 %02d/%02d/%04d",hour,minute,second,month,day,year))
		--else
		--	print("Unable to get time and date from the NTP server.")
		--end
	end
)
```

I'm hoping to make the writeup available on my website (www.limpkin.fr) in a month or so.

PR #1486 (2016-09-05): Make struct.unpack wrapped in setlist actually work
Fixes #1434.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This fixes the issue identified in the struct module (it reserves one too few entries on the stack). Also backported a fix from lua-5.2 to the op_setlist vm opcode so that the top of stack is not moved until all the entries have been copied off.

No documentation needs changing.

Committers supporting this PR: leave blan [... cut]
commit 99cd2177 (2016-09-05): Make struct.unpack wrapped in setlist actually work (#1486)

Fixes #1434.