nodemcu/nodemcu-firmware #1436: HSPI setup causes freeze

djphoenix, 2016-08-03:
### Expected behavior
HSPI set up and ready for communication

### Actual behavior
Console freeze & WDT reset

### Test code
Provide a minimal and isolated test which will reproduce the problem.
```Lua
spi.setup(
  1,
  spi.MASTER, 
  spi.CPOL_LOW, 
  spi.CPHA_HIGH, 
  spi.DATABITS_8,
  20,
  spi.HALFDUPLEX
)
```

### NodeMCU version
Branch: dev
Commit: `f1afe7b`

### Hardware
NodeMCU v2 AMICA

devsaurus, 2016-08-03:
> Console freeze & WDT reset

The test code works for me (Wemos D1 mini) - both with an SD card connected and without connections at all. I also don't see a possibility where the firmware would wait for an event to continue (and thus hang until WDT bites).

Do you have external components connected to the HSPI pins? GPIO15 is pulled down to GND with a 10k resistor, not directly?

djphoenix, 2016-08-03:
Pins are not connected at all, this is just a sample. Received report from our engineers team. I haven't real hardware to check pin connections. Just test on devkit.

marcelstoer, 2016-08-03:
Works for me, NodeMCU 1.0/v2 with nothing hooked up to the SPI pins.

djphoenix, 2016-08-20:
My miss. Flashed binary with SPI flash frequency "20m". Anyway not understood how it depends... But now it works.

closed 2016-08-20 by djphoenix