nodemcu/nodemcu-firmware #1443: Latest build bricking ESP-12E board

netomarin, 2016-08-07:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Flash working
### Actual behavior
After flashing with the downloaded build, the blue led starts blinking very fast and printing trash on terminal.

Selected the following 12 modules: cjson file gpio http i2c mqtt net node ow tmr uart wifi
Generated file: nodemcu-dev-12-modules-2016-08-07-19-19-16-integer.bin

Command used to flash:
esptool.py --port /dev/tty.wchusbserial1410 write_flash 0x00000 nodemcu-dev-12-modules-2016-08-07-19-19-16-integer.bin 

When I used the version 0.9.6, everything were ok. Information:
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
lua: cannot open init.lua
> =node.info()
0	9	6	14056186	1458400	4096	0	40000000
> =node.chipid()
14056186
> =node.flashid()
1458400

### Hardware
ESP-12E - NodeMCU Doit AM
8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-08-07:
> After flashing with the downloaded build, the blue led starts blinking very fast and printing trash on terminal.

On my modules this is a symptom of invalid init data. Have you tried the instructions from [upgrading firmware](http://nodemcu.readthedocs.io/en/master/en/flash/#upgrading-firmware)?

TransportLayer, 2016-08-07:
Can confirm issue with my ESP-01 board with release `1.5.4.1-master_20160802` and following the firmware upgrade instructions. Building and flashing release `1.5.1-master_20160603` will fix the board.

netomarin, 2016-08-07:
@devsaurus Before flashing a new firmware, I always erase flash completely with the command:
esptool.py --port /dev/tty.wchusbserial1410 erase_flash

@TransportLayer I'll try with release 1.5.1-master_20160603

Thanks

marcelstoer, 2016-08-07:
@netomarin As @devsaurus said this symptom indicates invalid init data or invalid `write_flash` parameters (I know it all too well but they're all in the docs). I just flashed another 12E module with a 1.5.4.1 firmware and all is well.

> esptool.py --baud 230400 --port /dev/tty.wchusbserial1410 write_flash -fm dio -fs 32m -ff 40m 0x00000 ~/Downloads/nodemcu.bin 0x3fc000 ~/Downloads/esp_init_data_default.bin

closed 2016-08-07 by marcelstoer

eku, 2016-08-08:
Remembers me of #1428.

netomarin, 2016-08-08:
It's working after added the esp_init_data_default.bin

Thanks

SimonFili, 2016-11-24:
Hi All,

Warning: Newbie post! be kind ;-)

I have a similar issue with an ESP-12e (4MB I think).  Here's what I did

This was the output with the old firmware:

> ESP12e
> 	NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4
> 	----------------------------
> 	dht22.lua       : 2719 bytes
> 	init.lua        : 112 bytes
> 	mainIoTF.lua    : 3909 bytes
> 	----------------------------
> 	Total file(s)   : 3
> 	Total size      : 6740 bytes
> 
> 	Total : 3441461 bytes
> 	Used  : 16566 bytes
> 	Remain: 3424895 bytes

1) Installed an old 0.9.x version on it and was able to connect and ran some code.
 - I got some not enough memory error on loading dht22 module... so decided to try a newer firmware
2) Went to nodemcu-build.com and build one with those modules:
> This was built against the master branch and includes the following modules: adc, bit, coap, dht, enduser_setup, file, gpio, http, i2c, mqtt, net, node, ow, pwm, rtcmem, rtctime, spi, tmr, uart, websocket, wifi, ws2801, ws2812.
3) Loaded the new firmware, rebooted and got bunch of garbage on the serial... same as what started that thread... So started investigating
4) Then I flashed/erase this module with a blank_1MB.bin to those location:
 0x000000, 0x100000, 0x200000, 0x2F00000 (I tried to 0x300000 but for some reason fails)
5) Re-flashed the custom firmware and get the same behavior.... so erasing (if I did correctly) did not fix my issue.

I could try what was proposed with esp_init_data_default.bin but I have no clue where to pick that file and to which location to flash it.
I am using the ESP8266Flasher windows GUI to flash.

Thanks for any hints to resolve this.

pjsg, 2016-11-24:
The current version uses a baud rate of 115200 rather than 9600 as was used in 0.9.6
---
Philip Gladstone. On the road, please excuse typos.

Loaded the new firmware, rebooted and got bunch of garbage on the serial... same as what started that thread... So started investigatingThen I flashed/erase this module with a blank_1MB.bin to those location:
0x000000, 0x100000, 0x200000, 0x2F00000 (I tried to 0x300000 but for some reason fails)Re-flashed the custom firmware and get the same behavior.... so erasing (if I did correctly) did not fix my issue.

SimonFili, 2016-11-24:
This is what I get with ESPlorer at 115kbit:

> PORT OPEN 115200
> 
> Communication with MCU..�oo�lorŽ’“nâ$`pòoàGot answer! AutoDetect firmware...
> 
> Can't autodetect firmware, because proper answer not received. Please, reset module or continue manually...
> �2NNŽrlNrŽ“›oâl`pòNàr„ÜÜâàboã|„ŒãŒŽp„ònnî�$`�NnŽ{$nsŽ›²oƒl pònàr„ÜÜâà„œboã|bŒòoNÎŽ

The leds on the uart board are madly blinked garbage.

marcelstoer, 2016-11-24:
> I could try what was proposed with esp_init_data_default.bin but I have no clue where to pick that file and to which location to flash it.

The email from the cloud builder even contains a link to http://nodemcu.readthedocs.io/en/master/en/flash/#upgrading-firmware