nodemcu/nodemcu-firmware #1446: FS compromised error

gbox3d, 2016-08-09:
### Expected behavior

### Actual behavior

Formatting file system. Please wait...

*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.

NodeMCU custom build by frightanic.com
        branch: master
        commit: 98d3b46e2c11068f361154a564ce75a048c9dadc
        SSL: false
        modules: cjson,file,gpio,net,node,pcm,struct,tmr,uart,wifi
 build  built on: 2016-08-09 02:28
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua

### Test code

flasing like below

./esptool.py --port /dev/tty.usbserial -b 230400 write_flash 0x00000 nodemcu-master-10-modules-2016-08-09-02-29-19-integer.bin 0x3fc000 esp_init_data_default.bin

### NodeMCU version

matser

### Hardware

esp-12

djphoenix, 2016-08-09:
I don't see specified flash mode/speed/size in command line. They're affects firmware configuration, so filesystem layout may be broken.

devsaurus, 2016-08-09:
Good catch @djphoenix.
@gbox3d what is the flash size of your module? Please add the [appropriate command line parameters](http://nodemcu.readthedocs.io/en/dev/en/flash/#esptoolpy):
- 4&nbsp;MByte: `-fs 32m -fm dio`
- 512&nbsp;kByte: `-fs 4m -fm qio` (and adjust the base address for esp_init_data_default.bin)

eku, 2016-08-09:
Since  #1442 the makefile contains rules with chip specific rules for flashing the firmware.

gbox3d, 2016-08-10:
try like below
./esptool.py --port /dev/tty.usbserial -b 230400 write_flash --flash_mode dio --flash_size 32m 0x00000 nodemcu-master-10-modules-2016-08-09-02-29-19-integer.bin 0x3fc000 esp_init_data_default.bin

but meet same message 

--------
*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.

NodeMCU custom build by frightanic.com
        branch: master
        commit: 98d3b46e2c11068f361154a564ce75a048c9dadc
        SSL: false
        modules: cjson,file,gpio,net,node,pcm,struct,tmr,uart,wifi
 build  built on: 2016-08-09 02:28
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
------

pjsg, 2016-08-10:
Can you dump the entire 4m of flash and upload it (use the read_flash command). I want to try flashing your exact contents and see if I get the same failure.

It *may* be the case that if the flash contained something bad before you flashed, then the format might fail. I don't think that that is the case, but it doesn't hurt to look.

After you dump the flash, try doing an erase_flash to clear the entire chip and then try the write_flash again and see if that solves the problem.

Please let us know what happens.

gbox3d, 2016-08-10:
today, I got  custombulid firmware at http://nodemcu-build.com/index.php 
and
retry 
./esptool.py --port /dev/tty.usbserial -b 230400 write_flash --flash_mode dio --flash_size 32m 0x00000 **nodemcu-master-9-modules-2016-08-10-00-57-30-integer.bin**
then success
this time , there is no FS compromised
there is just no differnce option esptool.py only differnce is yesterday's and today's firm

but i know flash size option is important

thank's a lot every one

closed 2016-08-10 by marcelstoer

Liriel, 2020-03-27:
fixe the same issue by calling

    esptool.py --port /dev/ttyUSB0 erase_flash

before re-flashing