nodemcu/nodemcu-firmware #1447: conn:send() behaviour with recent “NodeMCU custom builds” FW

arset, 2016-08-10:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
The code of the example “A simple http client” present on http://nodemcu.com/index_en.html 
should establish a connection with www.nodemcu.com and retrieve a small test web page.

### Actual behavior

The code of the example that used to work correctly with old FW (e.g.  nodemcu_integer_0.9.6-dev_20150704.bin)  run on ESP board flashed with new FW generated using “NodeMCU custom builds” doesn't return the expected web page. 

I got similar results replacing www.nodemcu.com 121.41.33.127 with the address of my website.
The code run on ESP board flashed with OLD FW works but the same code run on ESP board flashed with new FW doesn't unless I introduce a delay between the conn:connect() and the conn:send()

### Test code

-- A simple http client
conn=net.createConnection(net.TCP, false) 
conn:on("receive", function(conn, pl) print(pl) end)
conn:connect(80,"121.41.33.127")
conn:send("GET / HTTP/1.1\r\nHost: www.nodemcu.com\r\n"
    .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
I've repeater the test using two different ESP-12 and a ESP-1 with the same results.

### Remarks

The code of the example “A simple http client” present on http://nodemcu.com/index_en.html that used to work correctly with old FW (e.g.  nodemcu_integer_0.9.6-dev_20150704.bin)  apparently doesn't work anymore with new FW generated using “NodeMCU custom builds”.

I saved the code of the example:

-- A simple http client
conn=net.createConnection(net.TCP, false) 
conn:on("receive", function(conn, pl) print(pl) end)
conn:connect(80,"121.41.33.127")
conn:send("GET / HTTP/1.1\r\nHost: www.nodemcu.com\r\n"
    .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")

in a file; then I used “ESP8266 LuaLoader 0.91” to upload the file (“simple.lua”) on two of my ESP board that I have previously flashed using different FW: 
- nodemcu_integer_0.9.6-dev_20150704.bin (gotten from Git-Hub),
- nodemcu-master-7-modules-2016-04-09-17-13-19-integer.bin (generated last week using “NodeMCU custom builds” service).

Running dofile(“simple.lua”) by selecting “simple.lua” and pressing “dofile” button in “ESP8266 LuaLoader 0.91” terminal I got very different results depending from the flashed FW.

If I run the command on an ESP8266 flashed with the old FW ( nodemcu_integer_0.9.6-dev_20150704.bin), the ESP chip correctly connect to NodeMCU website, which return the expected html page;

but the same sequence executed on the ESP8266 flashed with the new FW (for instance nodemcu-master-7-modules-2016-04-09-17-13-19-integer.bin) produce NO result.

Repeating the test replacing  121.41.33.127  www.nodemcu.com with the address of a webserver under my control I have verified that “A simple http client” running under the new FW hang up immediately after establishing a “client connection”; that is the TCP/IP handshaking pass the accept() but the recv() never receive any data.

Doing some testing I have discovered that the problem seem in relation with the timing of the conn:send() because if I type “A simple http client” one row at a time sometime the code works.

The code also works with my website if I modify the code as follows:

-- A simple http client
conn=net.createConnection(net.TCP, false) 
conn:on("receive", function(conn, pl) print(pl) end)
conn:connect(80,"121.41.33.127")

sz0 = "GET / HTTP/1.1\r\nHost: www.nodemcu.com\r\n"
sz1 = "Connection: keep-alive\r\nAccept: */*\r\n\r\n"

tmr.alarm(0, 1000, 0, function()conn:send(sz0, sz1) end)

If you have any suggestion to fix the problem let me know, you are welcome.

Best regards

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-08-10:
The example is outdated and will not work reliably on current SDK/firmware versions, also see [`net.socket.send()`](http://nodemcu.readthedocs.io/en/dev/en/modules/net/#netsocketsend). There's a dedicated http client module available in the meantime with examples in [lua_examples/http-client.lua](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/http-client.lua)

marcelstoer, 2016-08-10:
Thanks for letting us know.

I updated the outdate examples on https://en.wikipedia.org/wiki/NodeMCU in June and asked @nodemcu (who is no longer maintaining here, see #1010) to do the same on nodemcu.com. Seems I forgot to ask him to update the HTTP client example as well. Will do that now and also add an example using the HTTP module to Wikipedia.

closed 2016-08-10 by marcelstoer

arset, 2016-08-10:
marcelstoer,
I have just finish to try the "A simple HTTP client"  example you have put on  https://en.wikipedia.org/wiki/NodeMCU.
It works!
 thank you