nodemcu/nodemcu-firmware #1449: 1.5.1.x firmware enters a crash loop even before eLua is started

remcoder, 2016-08-11:
### NodeMCU version
both 1.5.1 & 1.5.4.1 

### Hardware
Nodemcu v2 (4Mbit flash, flash manufacturer 0xe0 (LG semiconductor?))
I'm powering it from the usb port of my macbook (it can deliver 1A) or using my bench supply.

### Steps to reproduce
First I used the (awesome!) nodemcu-build.com to build the firmware with these modules: file, gpio, net, node, tmr, uart, wifi, ws2812. Then I flash the firmware like this:

    esptool.py write_flash  0x00000 nodemcu-master-8-modules-2016-08-10-17-47-53-integer.bin

Then I reset the module using it's button and the it enters a loop where it repeats to following:

     ets Jan  8 2013,rst cause:2, boot mode:(3,6)

	load 0x40100000, len 23988, room 16 
	tail 4
	chksum 0x7b
	load 0x3ffe8000, len 2184, room 4 
	tail 4
	chksum 0xcb
	load 0x3ffe8888, len 8, room 4 
	tail 4
	chksum 0x60
	csum 0x60
	rf_cal[0] !=0x05,is 0xFF

### Expected result
So I never see the eLua prompt which is what I expected. The above procedure used to work for me using older firmwar though. 

### Note
Things I tried with the same results:
 - different nodemcu board (but same hardware)
 - different settings for the flash mode / flash size
 - master / dev firmware
 - an older firmware from master based on 1.5.1 from 2/3 weeks ago
 - erasing flash before writing the firmware using `esptool.py erase_flash`
 - writing both the firmware and an init block using `esptool.py write_flash  0x00000 nodemcu-dev-8-modules-2016-07-31-14-40-26-integer.bin 0x7c000 esp_init_data_default.bin`

Actually, my theory is that the firmware is too large for a 4Mbit flash chip. Could this be? Is dat what the last line in the boot log means?:
`rf_cal[0] !=0x05,is 0xFF`

I understand this is might not be regular Nodemcu firmware issue but I didn't know what would be a better place to go to. If only I could get some hints where to look further as after a couple of nights of banging my head on this I'm out of options.

djphoenix, 2016-08-11:
Please, read [flash docs](http://nodemcu.readthedocs.io/en/dev/en/flash/).
You need to upload `esp_init_data_default.bin` AND specify flash size/mode in esptool options.

marcelstoer, 2016-08-11:
>  I didn't know what would be a better place to go to.

http://nodemcu.readthedocs.io/en/latest/en/support/

closed 2016-08-11 by marcelstoer

remcoder, 2016-08-13:
thx @djphoenix, indeed it works with the size at 32mbit. I was on the wrong track when Expressif's boot1.1 reported '4Mbit'. 😄

PhilETaylor, 2016-10-18:
Sometimes you just want to reach into the monitor and give someone a hug for leaving a note on an issue that fixes your brainfart and loop/lopp/lopp/loop nightmares :) :)

Thanks guys!

> python ./esptool.py --port=/dev/cu.SLAB_USBtoUART  write_flash  -fm=dio -fs=32m 0x3fc000 ~/Downloads/esp_init_data_default.bin

..... and I can finally go to bed!

marcelstoer, 2016-10-19:
@PhilETaylor Note that if you build a firmware from the `dev` branch you don't need the esp init data part any more, the firmware does that for you (will be on master in six weeks).

PhilETaylor, 2016-10-19:
Actually last night EVERY custombuild I ran with the cloud service failed - even if I just entered my email and clicked the build button with no changes to the selections. The bin file I eventually used was one built from master with the cloud service earlier in the week... Im thinking the custom build service might be broken at the moment...

marcelstoer, 2016-10-19:
@PhilETaylor that was because the Espressif server from which we download the SDK during the build was unavailable for several hours.

PhilETaylor, 2016-10-19:
No worries - thought I would mention it just in case :)