nodemcu/nodemcu-firmware #1452: write to files hangs ESP8266-01 module

kwis2, 2016-08-13:
8<------------------------ BUG REPORT -----------------------------------------
according to http://www.esp8266.com/viewtopic.php?f=23&t=11251&start=0

write a lot of data to files hangs modules. 

### Test code
https://github.com/kwis2/test_SPIFFS

### NodeMCU version
NodeMCU custom build by frightanic.com
branch: master
commit: 98d3b46e2c11068f361154a564ce75a048c9dadc
SSL: true
modules: adc,bit,file,gpio,http,i2c,net,node,ow,rtcfifo,rtcmem,rtctime,sntp,spi,tmr,uart,wifi
build built on: 2016-08-05 08:59
powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
### Hardware
tested on ESP8266-01 modules from different vendors / shops / versions

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-08-14:
I was able to reproduce the issue here:
- The file system can be filled up to the 20k threshold without problems.
- From this point on, files are deleted as new ones are created.
- This pattern works well for a couple of iterations (~70), but execution slows down more and more.
- For me the module was not responsive in the end and the file system was corrupted.

pjsg, 2016-08-14:
It turns out that there is a newer version of spiffs now available (0.3.8). Maybe the best thing is to move to that version first.....

pjsg, 2016-08-14:
However, when I run this as a test program in the spiffs test suite it works (actually it runs out of disk space). One of the solutions for this is to fake the 'used' space to be two BLOCKS bigger than spiffs reports. This will leave it some space to do the copy/erase stuff.

When I ran the test code on a large FS (about 3M), it didn't fail (with the new spiffs code).   I got to

```
curr file 2706 file to del 2207
awyn2707.dat
501
Total : 3322738 bytes
Used  : 629508 bytes
Remain: 2693230 bytes

curr file 2707 file to del 2208
```

before I gave up.....

kwis2, 2016-08-14:
From my observation problem start when you fill up whole memory - with 3M memory it could start with much more than 3300 files. to speed up try to change `maxFileNr = 5000` (meas.lua line 27) and size of single file to about 4k  -` for num=0, 400, 1 --init data file ` (meas.lua line 9)
the problem shpuld be visible abter about 1000 files (I guess)

kwis2, 2016-08-14:
sorry for closing - I am begginer on Github ;)

pjsg, 2016-08-14:
I'm wondering if the issue is really the filling up of the file system. Maybe one of the operations fails and the return code is not handled appropriately. When I tried the same code in the test harness, I got a file-system-full on the write call..... Maybe if I ignore this, things go badly wrong....

kwis2, 2016-08-14:
I don't know the SPIFFS concept, but in my opnion it could be problem with limited capacity of filesystem table, not filesystem capacity. Table could be not released properly. analogous behaviour could be possible with windows FAT, where entries are not cleared, but renamed. With limited capacity of FLASH it could be a problem.

kwis2, 2016-08-14:
On line `if(r<20000) then --less than 20k free` (meas.lua line 32) you could decide about amount of free space left, and with ESP8266-01 it doesn't matter if there is 20k or 100k free.

pjsg, 2016-08-15:
I found that if I increased the amount of free space required, then the code ran for a very long time and didn't fail. The amount of free space returned really is (AFAIK) the amount of free space (it is quite careful to remove all the internal overhead). It could be that the garbage collector can't run enough to collect the free space. I'm still looking at the underlying SPIFFS code.

kwis2, 2016-08-15:
there is a few more details I found, maybe it help in search.
Code I have sent it is small part of bigger one. the idea it was to write sensor data to file, sometimes HTTP post data, rename file with data which was sent, and delete files if there were no space.
The first symptom it was unable to rename files.
I have checked logs, and It looks that with 100k free space system hangs after 3.8k iterations (attempts to HTTP post not mean successful file writes) while it was unable to send post after about 2k
More interesting are results from test code with different file sizes with constant free space 20k (in fact 20 000B):
I have assumed that cluster it could be 256B, so I have tested multiplies of 256B - 1B. amazing is the repeatability of iteration to hangs when I repeat tests.
file size 8 x cluster (C) - 2047Bytes - 179 iterations (i), 30 deleted (d)
6 x C - 1535B - 222 i, 36 d
4 x C - 1023B - 318 i, 69 d
3 x C - 767B - 395 i, 97 d
2 x C - 511B - 456 i, 83 d
1 x C - 255B - 645 i, 147 d
Note number of iterations module alive strongly depends on file length. With no proportionality to free space, as we could assume.
767B file length work longer than 511B and 1023B (I have tested twice)
But when I run code with 127B file length I am not able to hang filesystem. Now it is 27k iterations and system still works

BTW. I have searched for SPIFFS documentation but I am not able to find any. could you give me any link?

kwis2, 2016-08-15:
with 127B file length there is most interesting behaviour 
I don't know what it exactly mean, but module crash with:
`curr file 68700 file to del 67953
awyn68701.dat
749
E:M 1032
PANIC: unprotected error in call to Lua API (not enough memory)

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 25952, room 16 
tail 0
chksum 0xe0
load 0x3ffe8000, len 2256, room 8 
tail 8
chksum 0xf3
load 0x3ffe88d0, len 8, room 0 
tail 8
chksum 0xbf
csum 0xbf
`

pjsg, 2016-08-16:
I opened an issue with the SPIFFS code: https://github.com/pellepl/spiffs/issues/99 -- I don't understand how the spiffs GC works....

pjsg, 2016-08-19:
It appears that when the amount of free space gets small, the spiffs gc can fail to create enough space to write data. 

In the example above, with the 127 byte file, the lua interpreter runs out of memory. 

I suspect that the root cause of all of this is that the GC gets slower and slower as the filesystem gets more and more fragmented. Eventually, the open fails, and then the write throws an exception.

pjsg, 2016-08-29:
This ticket caused me to do PR https://github.com/pellepl/spiffs/pull/100 which adds support for fuzzing the spiffs file system. It turns out that there are some subtle bugs which need to be resolved -- I don't know whether the issue that you are seeing is the same or not. 

In any case, the current SPIFFS code needs a fair amount of free space to operate correctly. I don't know what this size is......

closed 2019-06-21 by stale[bot]