nodemcu/nodemcu-firmware #1453: Bug in floor function

hwmland, 2016-08-13:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
floor (n) of negative integer should be n 

### Actual behavior
floor (n) of negative integer returns (n - 1)

### Test code
```Lua
print (math.floor (-1)) --returns -2 instead of -1
print (-2 % 2) --returns 2 instead of 0 because of wrong floor implementation
```
### NodeMCU version
Actual master (commit 98d3b46e2c11068f361154a564ce75a048c9dadc) (float firmware)
Problem is in file nodemcu-firmware/app/libc/c_math.c

### Recomended solution (not tested!)
```C
double floor(double x)
{
    return (double) (x < 0.f ? ((int)x == x ? x :  (((int) x) - 1)) : ((int) x));
}
```

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2016-08-13:
Thanks for the proposed fix, makes sense: #1454 .

jmattsson, 2016-08-19:
Fix merged into dev.

commit ced7ddc9 (2016-08-19): Fix math.floor for negative integers (#1454)

Fixes #1453
closed 2016-08-19 by jmattsson