nodemcu/nodemcu-firmware #1456: Sample code for MQTT causes esp8266 restarting like crazy(using new firmware)

ErfanThinker, 2016-08-14:
Hi, I used the firmware generated in cloud service and flashed my esp8266 with it. then tried to save this code to it:
```
function onSuccess(client)
    m:subscribe("/topic",0, function(client) print("subscribe success") end)
    m:publish("/topic","hello",0,0, function(client) print("sent") end)
end

-- init mqtt client with keepalive timer 120sec
m = mqtt.Client("clientid", 120, "user", "password")

-- setup Last Will and Testament (optional)
-- Broker will publish a message with qos = 0, retain = 0, data = "offline" 
-- to topic "/lwt" if client don't send keepalive packet
m:lwt("/lwt", "offline", 0, 0)

m:on("connect", function(client) print ("connected") end)
m:on("offline", function(client) print ("offline") end)

-- on publish message receive event
m:on("message", function(client, topic, data) 
  print(topic .. ":" ) 
  if data ~= nil then
    print(data)
  end
end)

-- for TLS: m:connect("192.168.11.118", secure-port, 1)
m:connect("192.168.11.118", 1883, 0, function(client) print("connected")
                                        onSuccess(client)
                                        end, 
                                     function(client, reason) print("failed reason: "..reason) end)

-- Calling subscribe/publish only makes sense once the connection
-- was successfully established. In a real-world application you want
-- move those into the 'connect' callback or make otherwise sure the 
-- connection was established.

-- subscribe topic with qos = 0
--m:subscribe("/topic",0, function(client) print("subscribe success") end)
-- publish a message with data = hello, QoS = 0, retain = 0

m:close();
-- you can call m:connect again
```
and my nodemcu started to restart at rate of 2-3  per sec. 
I managed to flash it by chance(barely) and tried to send this code again to my esp again and it didn't happen again, but then again I gave code to my pal and he tried to do the same with his nodemcu and phenomena happened to his device too.
Please help.

NicolSpies, 2016-08-14:
I have tested on my code and not experiencing the same. I think the problem is in your onSucess(client) function where publish is called before subscribed has completed.

djphoenix, 2016-08-14:
@NicolSpies nope. Publish message is *queued* to send after subscribe...
Tested this sample code, no issues found (latest dev with my patch merged). Try to rebuild firmware in cloud, it should just work.

ErfanThinker, 2016-08-14:
@NicolSpies  @djphoenix  I just built the firmware like 16 hours ago from [here](http://nodemcu-build.com/) and twice, as I said for the first time it caused the phenomena and after a second flash it went okay, but for my pal it caused the problem again. so it is causing the problem by chance or something I guess?

djphoenix, 2016-08-14:
> like 16 hours ago

So #1445 was merged 9 hours ago. Please try with latest build, it should be more stable for MQTT.

ErfanThinker, 2016-08-14:
@djphoenix I'll try to see if it'll do any good, thanks for help :)

hwmland, 2016-08-14:
Did you flash new esp_init_data_default.bin as well?

ErfanThinker, 2016-08-14:
@hwmland no, what is it for and where can I get it?

hwmland, 2016-08-14:
@ErfanThinker Try to read instructions [here](http://nodemcu.readthedocs.io/en/master/en/flash/)

marcelstoer, 2016-08-14:
See http://nodemcu.readthedocs.io/en/latest/en/support/ and #1010.

closed 2016-08-14 by marcelstoer

ErfanThinker, 2016-08-14:
@djphoenix tried latest build, the same happend again. don't know whats wrong.