nodemcu/nodemcu-firmware #1459: How do I connect to a "path"

gitnik, 2016-08-15:
I am not sure if this is a feature request or if that feature exists and I just can't find it.
What I want to do is connect to a domain plus a path.

>net.createConnection(16, 0):connect(80, "api.someurl.com/path");
> DNS retry 1!
> DNS retry 2!
> DNS retry 3!
> DNS retry 4!
> DNS Fail!

Is this something that you guys think is worth adding?

marcelstoer, 2016-08-15:
You get a path from a host after you connected to it. This is something for #1010 and http://nodemcu.readthedocs.io/en/latest/en/support/ though.

closed 2016-08-15 by marcelstoer

gitnik, 2016-08-15:
I don't really get what you're saying, could you show some code examples? To me this still sounds like a bug/feature request

jmattsson, 2016-08-19:
> What I want to do is connect to a domain plus a path.

This sentence does not make sense. One does not *connect* to anything but an IP address and a port tuple. There are no paths on the networking layer. Please refer to a TCP/IP primer to further your understanding, and consider the links Marcel provided.