nodemcu/nodemcu-firmware #1460: Support microsecond timers

JaapBraam, 2016-08-15:
The SDK also supports microsecond resolution timers but the timers in the tmr package only provide millisecond timers. The last parameter of the SDK's ets_timer_arm_new function can be used to select us (0) or ms (1) timer resolution. At the moment this is fixed to 1 in the tmr.c code.

Maybe it is possible to add an extra optional parameter to indicate a microsecond interval, while when not using this parameter the default millisecond resolution is used (maintaining compatibility).

### Justification
In my project I need 256us timers, 1ms timers are to slow...

### Workarounds
No known workarounds.

hwmland, 2016-08-16:
@JaapBraam I doubt LUA@nodemcu is fast enough to make it feasible.

JaapBraam, 2016-08-16:
I am using delay() now. It works for now, but it makes the program less responsive. Using timers would be a much better option.

djphoenix, 2016-08-16:
Maybe will make a test binary with ms resolution timers. So lua really will make trouble...

marcelstoer, 2016-08-17:
> Using timers would be a much better option because it will be event driven.

Indeed, [our docs](http://nodemcu.readthedocs.io/en/latest/en/modules/tmr/#tmrdelay) don't state it explicitly but I picked up somewhere that the delay shouldn't be more than 50us IIRC.

djphoenix, 2016-08-20:
So I tested microsecond timers, and cannot establish intervals less than 1000us (with compiled and stripped code and in 160MHz CPU mode I got something like 800us). Is this a case to provide new (mostly unusual) functionality?

JaapBraam, 2016-08-22:
I also tested nanosecond timers, and saw the same behavior.... That's not usable, and looks like a bug in the toolkit. Until that bug is fixed it is of no use to add it to the nodemcu firmware.

jmattsson, 2016-08-23:
We might be in a better position when (if?) we switch over to the RTOS-SDK. For Lua code there would of course not be any realtime guarantees, but the entire timer-setup is *much* nicer in the RTOS-SDK (speaking as a close-to-the-metal type person - upwards the stack it looks similar, of course).

djphoenix, 2016-08-23:
@jmattsson note that there is no nanosecond timer in RTOS SDK anyway ([ref](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/include/espressif/esp_timer.h)).

However RTOS is really better than NONOS. But this is another thread.

marcelstoer, 2016-09-01:
Not feasible ATM -> closing.

closed 2016-09-01 by marcelstoer

cyclingLinguist, 2020-09-29:
Is a microsecond timer feasible yet? I'm trying to use my ESP8266 to run a neopixel ring.