nodemcu/nodemcu-firmware #1461: Problem with 1.5.4 firmware

Phisherman, 2016-08-16:
Hi,

I'm able to flash the 1.5.4.1 dev branch firmware with the custom firmware builder to my NodeMCU. But if I connect to it with ESplorer via the serial port and 9600 baudrate i just get hieroglyphs. I tried the float and integer firmware with the same result. The 0.9.5 firmware works wonderful. Did I forget sth. to change? Sorry, I'm totally unexperienced in this area.

luismfonseca, 2016-08-16:
Have you [read flash docs](http://nodemcu.readthedocs.io/en/dev/en/flash/)?

I think the new version of the SDK (1.5.4.1) requires you to flash a new `esp_init_data_default.bin`.

Fetch it, then run something like (example for a 12E module): `esptool.py --baud 230400 --port /dev/tty.wchusbserial1410 write_flash -fm dio -fs 32m -ff 40m 0x00000 ~/Downloads/nodemcu.bin 0x3fc000 ~/Downloads/esp_init_data_default.bin`

Phisherman, 2016-08-16:
Thanks. I think it's working know. Indeed I didn't bestow consideration upon the flash docs.

closed 2016-08-16 by devsaurus

gregersrygg, 2016-08-21:
I followed the flash docs, but tried the “Erase flash completely” option. Experienced the same problem as described above - no matter the baud rate.

I can connect after flashing with the `esp_init_data_default.bin` file.

Did I erase the flash the wrong way, or are the docs wrong in that it's an option to erase the flash? I used this command before writing the new firmware:

```bash
$ esptool.py --port /dev/tty.wchusbserial1410 erase_flash
esptool.py v1.2-dev
Connecting...
Erasing flash (this may take a while)...
```
(no further output before it returned to the command line)

Phisherman, 2016-08-23:
@gregersrygg Erasing_flash erases the complete flash. As far as i understand it's optional. You just need to flash the new firmware and the `esp_init_data_default.bin`.

angellegna12, 2019-01-15:
You can download the firmware from  https://www.electrodragon.com/w/ESP8266_AT-Command_firmware