nodemcu/nodemcu-firmware #1462: Cant find the binary in the zip file.

adelima83, 2016-08-16:
Looking for the binary firmware on the master version. I cant seem to find it. All tutorials i have seen online seem to have a binary file but I cant find it. Any assistance is greatly appreciated.

luismfonseca, 2016-08-16:
Since there are many ways to build this project, I don't think there's an official `.bin`.

Read [this](https://nodemcu.readthedocs.io/en/dev/en/build/) to see how you can build yours. The [cloud service](http://nodemcu-build.com) is probably your easiest route.

adelima83, 2016-08-16:
Thank you. I was able to make a bin.

I have one more question for you. I flashed the new firmware but now i cant seem to connect to the normal baud that it was at previously. Do you know what the default baud rate is for NodeMCU vrs 1.5.4?

devsaurus, 2016-08-16:
Please see the [README.md](https://github.com/nodemcu/nodemcu-firmware#set-uart-bit-rate) and consider https://github.com/nodemcu/nodemcu-firmware/issues/1461#issuecomment-240257141.

closed 2016-08-16 by devsaurus