nodemcu/nodemcu-firmware #1463: wifi_event_monitor_handle_event_cb "queue not empty, appending queue"

kicer, 2016-08-17:
### Expected behavior

When chip try to connect router, will trigger wifi_event_monitor_handle_event_cb to handle event. as follows:

```
connected with Baidu3138, channel 11
dhcp client start...

    wifi_event_monitor_handle_event_cb is called

    queue empty, adding event and posting task
    	event 0

	STAMODE_CONNECTED
	Connected to SSID Baidu3138, Channel 11
[DEBUG] wifi.eventmon.STA_CONNECTED
```

### Actual behavior
  
Restart device for many times, will report "not empty" sometimes, but report "not empty" always after that.
  
```
connected with Baidu3138, channel 11         
dhcp client start...         

	wifi_event_monitor_handle_event_cb is called             

	queue not empty, appending queue
```

### Test code

```Lua
wifi.sta.eventMonReg(wifi.STA_IDLE, function() print("STATION_IDLE") end)
wifi.sta.eventMonReg(wifi.STA_CONNECTING, function() print("STATION_CONNECTING") end)
wifi.sta.eventMonReg(wifi.STA_WRONGPWD, function() print("STATION_WRONG_PASSWORD") end)
wifi.sta.eventMonReg(wifi.STA_GOTIP, function() print("STATION_GOT_IP") end)
wifi.sta.eventMonReg(wifi.STA_FAIL, function() print("STATION_CONNECT_FAIL") end)
wifi.sta.eventMonStart()
```

After register the eventMon, log as follows:

```
STATION_GOT_IP
state: 5 -> 2 (4a0)
rm 0
pm close 7

    wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
reconnect
state: 2 -> 0 (0)
STATION_CONNECTING
scandone
no Baidu3138 found, reconnect after 1s

	wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
reconnect
scandone
no Baidu3138 found, reconnect after 1s

	wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
reconnect
scandone
no Baidu3138 found, reconnect after 1s

	wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
reconnect
scandone
state: 0 -> 2 (b0)
STATION_CONNECTING
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Baidu3138, channel 11
dhcp client start...

	wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
ip:192.168.95.3,mask:255.255.255.0,gw:192.168.95.1

	wifi_event_monitor_handle_event_cb is called

	queue not empty, appending queue
STATION_GOT_IP
```

### NodeMCU version
master

### Hardware
ESP12-F module

dnc40085, 2016-08-18:
Hello, The `wifi.sta.eventmonxxx` functions are completely separate from `wifi.eventmon.xxx` functions.

Please refer to the documentation for [`wifi.eventmon.register`](http://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifieventmonregister) to see an example of how to use the wifi event monitor.

kicer, 2016-08-18:
@dnc40085 , tks. Yes, the `NODE_DBG` has defined in `app/include/user_config.h`.

Always report  `queue not empty` sometimes after boot is my doubt. With multi restart, `queue empty, adding event and posting task` seems right debug report.

dnc40085, 2016-08-18:
Ok, I'll take a look at it

dnc40085, 2016-08-18:
I tried a build of the current master branch with debugging turned on and I couldn't reproduce this bug, as far as I can tell, the event queue/dequeue is working properly.

Are you using a current build of the master branch?
can you post code that can reliably reproduce this bug?

kicer, 2016-08-22:
Sorry, I can't reliably reproduce this bug, it‘s a completely random~

The `task_count` variable not init, maybe this is the reason why task not be called by `task_dispatch() -> wifi_event_monitor_process_event_queue()`? 
```
app/task/task.c L23:
LOCAL int task_count;
```

dnc40085, 2016-08-23:
I can't tell from your original post what code you're actually using that may be triggering this issue.
Can you post your full test code?

kicer, 2016-08-24:
In fact, I has add some features in master, the original lua test code can't be restore~

After init the `task_count` variable as shown above, "`queue not empty, appending queue`" report not found now.

Thanks for your help, and I will test continues and feedback if that occur again.

marcelstoer, 2016-08-28:
@kicer feel free to re-open once you have a [verifiable example](http://stackoverflow.com/help/mcve) -> closing.

closed 2016-08-28 by marcelstoer

dnc40085, 2016-08-28:
@marcelstoer IMHO, it might be useful to include that link in `ISSUE_TEMPLATE.md`

marcelstoer, 2016-08-28:
Good idea, thanks. I added it. It'll become active on GitHub with the next master drop.

kicer, 2016-08-29:
@dnc40085, @marcelstoer , Sorry for my unclearly statements.

 it‘s a completely random(even <1/20)~ is likely to means that some memory is not initialized. I think that's why master branch couldn't reproduce this issue.  
Follow the code, may the `task_dispatch()` won't work as expect. so, please check the `app/task/task.c L21/L23` code, `task_Q/task_count` variables not initialized, may be a pull request need to make.

jmattsson, 2016-08-29:
Those variables are statics, and as such implicitly zero-initialized.

kicer, 2016-08-29:
@jmattsson, Yes, auto zero-initialized orz~. Thanks for your knowledge. 
I will test continues~~~ and feedback if that occur again.