nodemcu/nodemcu-firmware #1468: Unable to boot new Firmware

ludovicterrier, 2016-08-23:
Hello,

I'm unable to boot new firmware built from https://nodemcu-build.com/

The one built built on: 2016-07-09 08:33 works, but ones built today and yesterday don't. (I did many attempts with master/dev and 10, 11, 13, modules...)

I use esptool.py and I have no errors. But when I restart the module (Wemos D1 V2 or Huzzah Feather) the LED blinks very quickly. And that's all.

### NodeMCU version that's working
NodeMCU custom build by frightanic.com
        branch: master
        commit: b580bfe79e6e73020c2bd7cd92a6afe01a8bc867
        SSL: false
        modules: adc,bme280,file,gpio,mqtt,net,node,ow,pwm,tmr,uart,wifi,ws2812l.

### Hardware
Wemos D1 V2
& Huzzah Feather Adafruit

hwmland, 2016-08-23:
Sounds like mantra: did you flash new esp_init_data_default.bin as well? See [http://nodemcu.readthedocs.io/en/master/en/flash/](url)

marcelstoer, 2016-08-24:
See https://nodemcu.readthedocs.io/en/latest/en/support/ (also referenced on https://nodemcu-build.com/faq.php#get-help) and #1010.

closed 2016-08-24 by marcelstoer