nodemcu/nodemcu-firmware #1471: bmp085 module returns incorrect values (bmp180 sensor)

CosmicMac, 2016-08-24:
### Expected behavior
Right values returned by [bmp180 lua module](https://github.com/javieryanez/nodemcu-modules/tree/master/bmp180):
>Lua module temperature:    25.2
>Lua module pressure:    1009.9

### Actual behavior

Wrong values returned by bmp085 native module:
>Native module temperature:    143.9
>Native module pressure:    1344.5

### Test code
```Lua
local SDA_PIN = 4 -- BMP180 SDA pin on GPIO2
local SCL_PIN = 3 -- BMP180 SCL pin on GPIO0
local OSS = 3 -- BMP180 oversampling setting (0: ultra low power, 1: standard, 2: high resolution, 3: ultra high resolution)

-- Test with bmp180 lua module (https://github.com/javieryanez/nodemcu-modules/tree/master/bmp180)
local bmp180 = require("bmp180")
bmp180.init(SDA_PIN, SCL_PIN)
bmp180.read(OSS)
local t1 = string.format("%.1f", bmp180.getTemperature() / 10)
local p1 = string.format("%.1f", bmp180.getPressure() / 100)
bmp180 = nil
package.loaded["bmp180"] = nil

print("Lua module temperature:", t1)
print("Lua module pressure:", p1)

-- Test with bmp085 native module

bmp085.init(SDA_PIN, SCL_PIN)
local t2 = string.format("%.1f", bmp085.temperature() / 10)
local p2 = string.format("%.1f", bmp085.pressure(OSS) / 100)

print("Native module temperature:", t2)
print("Native module pressure:", p2)
```
### NodeMCU version
>NodeMCU custom build by frightanic.com
>    branch: master
>    commit: 95e85c74e7310a595aa6bd57dbbc69ec3516e9c6
>    SSL: false
>    modules: adc,bmp085,cjson,dht,file,gpio,i2c,net,node,tmr,uart,wifi
> build     built on: 2016-08-23 18:24
> powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
[ESP8266 board based on ESP-12f](http://www.banggood.com/3Pcs-Mini-NodeMCU-ESP8266-WIFI-Development-Board-Based-On-ESP-12F-p-1058306.html)

jmattsson, 2016-08-27:
Could you please have a look at this, @kbeckmann ?

igrcic, 2017-07-01:
Hi, 

i can confirm this on my Wemos Mini devici.

I am receiving the wrong value for the temperature reading, pressure seems to be  closer to the real value.

closed 2019-07-15 by stale[bot]