nodemcu/nodemcu-firmware #1473: wifi.setmode(wifi.NULLMODE) does not turn off WiFi

joysfera, 2016-08-24:
### Expected behavior
With WiFi turned off the ESP8266 should require less power (12 mA instead of 70 mA).

### Actual behavior
The measured current stays the same (still 70 mA) after calling ```wifi.setmode(wifi.NULLMODE)```. I suspect the WiFi was not turned off.

### Test code
```Lua
wifi.setmode(wifi.NULLMODE)
```
### NodeMCU version
Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
Plain ESP8266.

### BTW
There's a workaround for turning off WiFi by falling into dsleep(): ```node.dsleep(1, 4)```
This is rather inconvenient, though, as it requires GPIO16 to be wired to RST and it also resets the whole chip so your Lua program gets interrupted and starts again. It would be better if the ```wifi.setmode()``` actually worked as advertised and turned off WiFi without rebooting the whole chip.
Please note that this could be a limitation of SDK or even the hardware itself. I have yet to check that.

dnc40085, 2016-08-24:
This is not a bug. Auto sleep in null mode is not default functionality, PR #1231 will add this functionality once I can work out the kinks, which is soon I hope.

closed 2016-08-25 by marcelstoer

joysfera, 2016-08-25:
@dnc40085 thank you, looking forward for PR #1231

dnc40085, 2016-08-26:
@joysfera @marcelstoer Here is a [PR](https://github.com/nodemcu/nodemcu-firmware/pull/1475) to add this functionality