nodemcu/nodemcu-firmware #1474: recent firmwares very actively print garbage on the console

Habbie, 2016-08-26:
### Expected behavior
Powering the device quickly gives me a Lua prompt.

### Actual behavior
Console is filled with garbage, speed apparently mostly dependent on the baudrate I pick. Blue LED on Nodemcu board also blinks actively.

### NodeMCU version
Last known good: [6bb6996d2b99c51ff1664e43c31ace07ae168f5b] Fix GitHub source code link in docs for 'latest' path
First known bad: [84c005e85c78b4382e68633ab61e2f4928c838f5] Update to SDK 1.5.4.1 (#1390)

However, I cannot build the 13 commits before that, i.e. [549f42107ee8200b2016403d2d45295c714f7875] Update to SDK 1.5.4 (#1377) through [8866297c6d7cc35afd25b0f0f666ef90af040300] Remove buffer:write() as ws2812.write() handles buffers (#1408), they fail with:
```
(cd "/opt/nodemcu-firmware/sdk/" && rm -fr esp_iot_sdk_v1.5.4 ESP8266_NONOS_SDK && unzip /opt/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4*.zip ESP8266_NONOS_SDK/lib/* ESP8266_NONOS_SDK/ld/eagle.rom.addr.v6.ld ESP8266_NONOS_SDK/include/* )
Archive:  /opt/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4.1_patch_20160704.zip
caution: filename not matched:  /opt/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip
caution: filename not matched:  ESP8266_NONOS_SDK/lib/*
caution: filename not matched:  ESP8266_NONOS_SDK/ld/eagle.rom.addr.v6.ld
caution: filename not matched:  ESP8266_NONOS_SDK/include/*
Makefile:189: recipe for target '/opt/nodemcu-firmware/sdk/.extracted-1.5.4' failed
make: *** [/opt/nodemcu-firmware/sdk/.extracted-1.5.4] Error 11
```

(I did not investigate this failure yet).

Current dev and master heads are also bad.

In short, the 1.5.4 upgrade commit breaks building for me; the 1.5.4.1 upgrade commit unbreaks building but yields useless firmware.

### Hardware
Bisect done on NodeMCU board. Last good and first known bad commits also tested on Wemos D1 Mini with the same result.

devsaurus, 2016-08-26:
Please see #1468 et.al. for the correct firnware upgrade procedure.
To build 1.5.4 you need to remove the 1.5.4.1 zip files from `cache` folder. Makefile was too greedy with wildcard matching back then.

Habbie, 2016-08-26:
I did an `esptool erase_flash` before every `write_flash`, it was my understanding this is a correct upgrade procedure. Emptying the cache helps for the unbuildables, trying those now.

Habbie, 2016-08-26:
Re-bisected with working 1.5.4 builds, updated original text.

VSpectrum, 2016-08-26:
Following #1468 , I flashed the esp_init_data_default.bin. Although I was using an ESP-12E (4 MB) flashing the init_data to 0x3fc000 resulted in the same problem (endless blinking after reset and garbage output). However flashing to address 0x7c000 seemed to work for me:
 
esptool.py --port /dev/ttyUSB0 write_flash 0x00000 nodemcu-master-10-modules-2016-08-25-16-25-29-float.bin 0x7c000 esp_init_data_default.bin 

Is the note made concerning the addresses to flash to in http://nodemcu.readthedocs.io/en/master/en/flash/ confirmed?

devsaurus, 2016-08-26:
Yes, the addresses are working, given you configured the flash correctly.
Please also specify the command line parameters `-fm` and `-fs` as mentioned there.

VSpectrum, 2016-08-26:
```sh 
$ esptool.py --port /dev/ttyUSB0 write_flash 0x00000 nodemcu-master-10-modules-2016-08-25-16-25-29-float.bin 0x3fc000 esp_init_data_default.bin -fm dio -fs 32m 
```
Adding and specifying the flashmode and flashsize flags worked for me, thanks.

Habbie, 2016-08-26:
`-fm dio -fs 32m` also work on my NodeMCU. Apparently these things have become stricter with SDK 1.5.4.1. Thank you for your patience, @devsaurus!

closed 2016-08-26 by Habbie

rohit-gandhe, 2016-10-19:
Where do I get esp_init_data_default.bin? I don't see it in my bin dir after the build.

rohit-gandhe, 2016-10-19:
Never mind, downloaded as part of the SDK