nodemcu/nodemcu-firmware #1476: gpio.trig fastest edge detection is 2ms.  How to speed up edge detection to micro seconds?

danlu01, 2016-08-27:
I'm trying to detect the high pulse of a signal.  However, gpio.trig can only detect up to 2ms pulse after setting CPU freq to 160mhz.  Is there a way to detect pulses in the micro seconds range?

In my experiment, I common the gnd between ESP8266-ESP12 and Arduino mega.  A digital output pin of the arduino is connected to pin 1 on the ESP8266.  Arduino loops with switch the output pin to create a high pulse.  When I shorten the high pulse, ESP8266 can only detect down to 2ms.

djphoenix, 2016-08-27:
Welcome to Lua world. In plain C firmware I achieved 104us-precision interrupt handler (enough for 9600bps UART RX). But Lua have some restrictions, and too slow...

marcelstoer, 2016-08-27:
See #1010 and http://nodemcu.readthedocs.io/en/latest/en/support/.

closed 2016-08-27 by marcelstoer