nodemcu/nodemcu-firmware #1479: spiffs image format failed when first boot

kicer, 2016-08-28:
### Actual behavior
I have enabled the `DEVELOP_VERSION` macro, then build and got the `0x90000-32mb.bin` file.
After download firmware(nodemcu*.bin -> 0), then restart device, the log shown as follows
* fs.start is 0x90000 or 0xa0000 ?

```
softAP(1a:fe:34:cc:85:ec)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

_flash_used_end:40283d7b
fs.start:90000,max:36a000
mount res: -10025, -10025
Formatting file system. Please wait...
_flash_used_end:40283d7b
fs.start:a0000,max:35a000
mount res: 0, 0
Formatting: size 0x35a000, addr 0xa0000
_flash_used_end:40283d7b
fs.start:90000,max:36a000
mount res: -10025, -10025

*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.
Task task_lua started.

NodeMCU 1.5.4.1 build Aug 26 2016 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
> Heap size::33800.
> =file.fscfg()
589824	3579904
> 
=file.fsinfo()
stdin:1: file system failed
stack traceback:
	[C]: in function 'fsinfo'
	stdin:1: in main chunk
> =file.format()
_flash_used_end:40283d7b
fs.start:a0000,max:35a000
mount res: 0, 0
Formatting: size 0x35a000, addr 0xa0000
_flash_used_end:40283d7b
fs.start:90000,max:36a000
mount res: -10025, -10025

*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.
stdin:1: Failed to format file system
stack traceback:
	[C]: in function 'format'
	stdin:1: in main chunk
```
### NodeMCU version
master, but with some features add~

### Hardware
ESP-12F module

pjsg, 2016-08-28:
Did you flash the 0x90000-32mb file to the ESP? It should go at offset 0x90000... 

The firmware tries to find a good image by looking at various offsets -- I'm not sure what is happening here. I'l stare at the code tomorrow to see what that sequence of output really means....

kicer, 2016-08-28:
No, not flash the 0x90000-32mb file. I run objdump as `tools/Makefile`, the result as this:

```
(0x`$(OBJDUMP) -t ../app/.output/eagle/debug/image/eagle.app.v6.out |grep _flash_used_end |cut -f1 -d" "` - 0x40200000)

0x40283d7b - 0x40200000 = 0x83D7B
```

kicer, 2016-08-28:
Another issue:
In file `app/include/user_config.h`, we can use `SPIFFS_FIXED_LOCATION` and `SPIFFS_SIZE_1M_BOUNDARY` to set the fs start address.
but the spiffsimg start address build by `tools/Makefile->FLASH_USED_END` not care that marco.

kicer, 2016-08-28:
Flash the 0x90000-32mb file and firmware:
```
softAP(1a:fe:34:cc:85:ec)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

_flash_used_end:40283d7b
fs.start:90000,max:36a000
mount res: -10025, -10025
Formatting file system. Please wait...
_flash_used_end:40283d7b
fs.start:a0000,max:35a000
mount res: -10025, -10025
Formatting: size 0x35a000, addr 0xa0000
_flash_used_end:40283d7b
fs.start:90000,max:36a000
mount res: -10025, -10025

*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.
Task task_lua started.

NodeMCU 1.5.4.1 build Aug 28 2016 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
> Heap size::34112.
```

Flash the 0x90000-8mb file and firmware:
```
softAP(1a:fe:34:cc:85:ec)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

_flash_used_end:40283d7b
fs.start:90000,max:36a000
Overriding size:6c000
mount res: 0, 0
Task task_lua started.

NodeMCU 1.5.4.1 build Aug 28 2016 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
bcn 0
del if1
usl
mode : sta(18:fe:34:cc:85:ec)
add if0
> Heap size::30824.

...... init.lua debug msg, remove it~

> =file.fscfg()
589824	442368
> =file.fsinfo()
total: 404863, used:13554
391309	13554	404863
> =file.format()
_flash_used_end:40283d7b
fs.start:a0000,max:35a000
mount res: -10025, -10025
Formatting: size 0x35a000, addr 0xa0000
_flash_used_end:40283d7b
fs.start:90000,max:36a000
Overriding size:6c000
mount res: -10025, -10025

*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.
stdin:1: Failed to format file system
stack traceback:
	[C]: in function 'format'
	stdin:1: in main chunk
```

pjsg, 2016-08-29:
When you flash the firmware, do you specify --flash_size 32m (actually, what size flash chip do you have?)

It looks as though there is some confusion as to the size of the flash chip -- some operations think that the size is 4MB and some think it is 1MB.

kicer, 2016-08-29:
I use `ESPFlashDownloadTool_v3.3.4` for download in win10, just select chip size is 32mb.
The module is ESP-12F that flash size is 4MB. refer the documents of spiffs,allowed flash a small image in 4MB chip. I just want to verify spiffs-image file is valid or not.

> The disk image file is placed into the bin directory and it is named 0x<offset>-<size>.bin where the offset is the location where it should be flashed, and the size is the size of the flash part. It is quite valid (and quicker) to flash the 512k image into a 4M part. However, there will probably be limited space in the file system for creating new files.

Seems when call the `file.format()`, `fs_start` is incorrect(0x90000->0xa0000).

pjsg, 2016-08-30:
The current code tries to find the current filesystem by searching for it. It turns out that it will sometimes recognize a filesystem in the wrong place (but it is much better than the old code).

MitchBradley, 2016-10-05:
I just ran into the "recognize FS in the wrong place" issue, after having flashed new firmware that was larger than the old.  That resulted in corruption of some of the filesystem data, so I tried to fix it by calling myspiffs_format().  That gave "Formatting: size 0x35a000, addr 0xa0000", but then when it tried to remount the FS, size/addr was 0x36a000/0x90000.  There was still a residual file in the chunk at offset 0x90000.

I fixed it, I think, by changing (in myspiffs_set_cfg())

`    myspiffs_set_cfg(cfg, 0x10000, 0x10000, TRUE);
`
to 
`    myspiffs_set_cfg(cfg, 0x10000, 0, TRUE);
`

That solved the problem; re-running myspiffs_format() resulted in a pristine file system that seemed to behave correctly thereafter.

It would appear that the 0x10000 offset is incorrect, but perhaps there is some subtlety that I am missing.

pjsg, 2016-10-05:
Ugh. It turned out that the way that spiffs marks the file system makes it impossible to figure out where the start is.... I have some patches that would change the filesystem layout, but would, I think, solve this problem.

The spiffs author is going to change the format in the future to improve the performance when the filesystem starts filling up. This issue will be fixed then. However, it might take some time....

Should we just change the filesystem format (a little bit) to fix this?

jmattsson, 2016-10-05:
Considering that people have run into issues in this area, I'd say we probably should. It shouldn't cause us any more issue than we'll get anyway when the on-disk format changes down the track, should it?

pjsg, 2016-10-05:
I have code hat should resolve this problem, but it will trigger a reformat.....

jmattsson, 2016-10-05:
Do we have an established way of getting that fact into the release notes to give users a heads-up? The reformat, I mean.

pjsg, 2016-10-05:
We could say that the file system is preserved if possible, but will sometimes be destroyed when reflashing firmware. It is designed to reduce the chance of requiring a reformat when flashing firmware of similar versions and similar size. 

The precise set of rules is quite complex and can be affected by various options at compile time....

marcelstoer, 2016-10-05:
> Do we have an established way of getting that fact into the release notes to give users a heads-up?

There is but I wouldn't consider it "established". With the last few releases I kept tracking changes in the [milestone](https://github.com/nodemcu/nodemcu-firmware/milestones) (to which I added merged PRs). When I eventually cut the [release](https://github.com/nodemcu/nodemcu-firmware/releases) I simply transferred the description and elaborated on issues I remembered.
I made a mental note a while ago that I want to document for us internally how releases are done.

MitchBradley, 2016-10-05:
Is the 0x10000 (vs. 0) in the "set up for a format" clause a bug?  It seems inconsistent to format starting at offset 0x10000 but to scan from offset 0.

pjsg, 2016-10-05:
That is deliberate. When formatting, we leave a space between the end of the firmware and the start of the file system. This is so, if the firmware grows in size, then it is unlikely that it will overwrite the start of the file system. This is also why it searches so many locations.

The bug is that the spiffs code, when asked if a particular offset is a valid file system, returns YES even if the pointer is not to the start of a filesystem...

MitchBradley, 2016-10-05:
That being the case, shouldn't the scanning code also respect the offset, by starting the search at index 1 instead of index 0 in the following?

`
    if (INTERNAL_FLASH_SIZE >= 700000) {
      for (i = 0; i < 8; i++) {
`

I'm trying to think of scenarios where it is useful to scan the preceding block, but I keep coming up with pathologies that negate the possible advantages.

pjsg, 2016-10-05:
You want to scan the preceding block if the next firmware is larger than the current firmware (as then the next block will be the first block -- if you understand what I mean). The real problem is that you can end up with multiple file system headers in memory and choosing between them is difficult/impossible. 

The new code makes sure that we are actually pointing at a filesystem header, *and* the last block of the file system is the last block of that filesystem.

MitchBradley, 2016-10-05:
Yep, that's the sort of pathology I was thinking of, which I already hit.

Your new code sounds good.

MitchBradley, 2016-10-05:
Hmmm.   Another safety measure occurred to me.  There are two ways to reach the "set up for a format" code - either via force_create or by falling through the !force_create clause.  They could (should?) be handled differently.

In the force_create case, it is possible that scanning would find something.  If it does, and if the resulting phys_addr differs from that found by myspiffs_set_cfg(cfg, 0x10000, 0x10000, TRUE), it means that there are fragments that probably should be erased.

Another way to accomplish the effect of cleaning out junk, with less decision making, would be to erase 4K sectors after platform_flash_get_first_free_block_address().

Please tell me if you are tired of my kibitzing :-)

PR #1528 (2016-10-25): Improve reliability of FS detection.
Fixes #1479 (at least the most recent part)

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This gives each block in the FS a magic number such that the difference between adjacent blocks tells you whether you are at the start of the FS or not.