nodemcu/nodemcu-firmware #1480: wifi.sta.disconnect() does not sleep the modem

joysfera, 2016-08-28:
### Expected behavior
when wifi is disconnected it would be logical if the wifi modem went to sleep. Especially when the sleeptype is set to MODEM_SLEEP.

### Actual behavior
After disconnecting the power load increases up to 70 mA - that indicates the wifi modem is fully powered.

### Test code
Provide a minimal and isolated test which will reproduce the problem.
```Lua
wifi.setmode(wifi.STATION)
wifi.sleeptype(wifi.MODEM_SLEEP)
wifi.config(...,...)
wifi.connect()
-- here the ESP needs about 15 mA
wifi.sta.disconnect()
-- now the ESP requires more than 70 mA
```
### NodeMCU version
custom build by frightanic.com
	branch: dev
	commit: 3328c66f2c1c4b83f96b3c869b2167bb2ab183a2

### Hardware
plain ESP8266 version 07

Still trying to figure out how to program a battery powered ESP device. It would be logical to connect to an AP only when needed, fetch fresh data, then disconnect immediately and run in low power. However, currently disconnecting from an AP actually increases battery load. Hopefully can be fixed similarly to the recent setmode(wifi.NULLMODE) fix.

That's actually the only way now to save power - i.e. change the mode from STATION to NULLMODE after disconnecting and then change the mode back before reconnecting. I'd think it shouldn't be necessary if the MODEM_SLEEP worked in all cases hence I'm submitting this issue. Thanks!

dnc40085, 2016-08-28:
[ESP8266 - Low Power Solutions(PDF 509KB):](http://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf)
>Modem-sleep mode is enabled only when ESP8266 connects to the router in station 
mode.

This is a limitation of the SDK. In order to go into modem sleep when disconnected from AP `wifi.setmode()` must be set to `wifi.NULLMODE` otherwise wifi modem will stay awake.

It would be possible to change the code for `wifi.sta.disconnect` to this automatically, but I think it would blur the description of `wifi.sta.disconnect()`.

If others think it would be beneficial to add this feature, I have no problem writing a PR to implement this.

joysfera, 2016-08-28:
Ah, thanks for the explanation @dnc40085 
I would vote for mentioning it in the NodeMCU documentation rather than modifying the ```wifi.sta.disconnect()``` behavior. Something along the lines: "if you want to save power call the ```wifi.setmode(wifi.NULLMODE)```, disconnecting itself will not turn the modem off as one would expect."

marcelstoer, 2016-08-28:
@joysfera @dnc40085 👍 for docs addendum, PRs welcome 😉

dnc40085, 2016-08-28:
>Although there's an unnecessary power spike... I'm wondering if it is OK to call setmode(wifi.NULLMODE) when you want to disconnect from an AP.

The power spike is most likely caused from the sdk function wifi_set_opmode writing mode to flash, I've got a [PR](https://github.com/nodemcu/nodemcu-firmware/issues/1288) in the works to allow changing opmode without writing to flash.

As for using `wifi.setmode()` instead of `wifi.sta.disconnect()` to disconnect from AP. As far as I can tell, there is no problem using `wifi.setmode()` to disconnect from AP, I would do some testing to be sure though.

joysfera, 2016-08-28:
@dnc40085 I thought the power spike coming from the wifi modem being fully powered. The ```wifi.sta.disconnect()``` call ends the sleeptype power saving feature (as you've just shown me in the SDK DOC) thus the modem goes to full power immediately. Then Lua processes the next program line that does ```wifi.setmode(wifi.NULLMODE)``` and this turns the modem off again. But the moment between disconnecting and changing the mode the power spike appears (I believe, not really tested on a scope). That's why I was thinking about calling the ```setmode(NULLMODE)``` even before disconnecting from an AP thus hopefully avoiding the power spike.

dnc40085, 2016-08-28:
>I thought the power spike coming from the wifi modem being fully powered

I was under the assumption that you were saying that you were getting a power spike when using `wifi.setmode(wifi.NULLMODE)` instead of `wifi.sta.disconnect()` to disconnect from the AP.

Under this assumption, I did some testing to determine if the power spike is caused by wifi disconnection or write to flash by `wifi.setmode()` and I found that a power spike was caused by writing wifi mode to flash.

FWIW I tested with ammeter

joysfera, 2016-08-28:
@dnc40085 no, I said I would try using the ```NULLMODE``` instead of ```disconnect``` because the disconnect causes the power spike. Actually it's does not cause spike - the power simply goes up to the usual 70+ mA and would stay there. But the subsequent call to ```setmode(NULLMODE)``` lowers the power again so the result is a spike.

dnc40085, 2016-08-29:
my mistake

joysfera, 2016-08-29:
@dnc40085 BTW, when ```wifi.setmode()``` writes to flash, will not it kill the ESP device in the long run by wearing out its flash memory? Say I change the mode 24 times per minute, that's 35 thousand per day, over 12 millions per year. That sounds like a killer.

dnc40085, 2016-08-29:
If you are setting the same mode over and over, no. But if you are switching modes, I believe it does write to flash every time. 
To remedy this problem, I am currently working on a PR to make writing mode to flash optional.

dnc40085, 2016-11-07:
@marcelstoer since PR #1497 has been merged, I believe this issue can be closed

closed 2016-11-07 by marcelstoer