nodemcu/nodemcu-firmware #1481: crash in

joysfera, 2016-08-28:
I think I'm experiencing a crash inside of NodeMCU firmware after calling ```wifi.setmode(wifi.NULLMODE)``` - this happens only when called after wake up from ```rtctime.dsleep(xxx, 4)```. I don't know if I managed to isolate the case properly (I'm editing a larger source code) but it seems so.

### Expected behavior
no crash

### Actual behavior
crash: (please see the unedited serial console output below)

```
Starting
bootreason: 2, 6
Waking up from sleep
Calling wifi.setmode(wifi.NULLMODE)
> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 24368, room 16 
tail 0
chksum 0xc3
load 0x3ffe8000, len 2324, room 8 
tail 12
chksum 0xd6
ho 0 tail 12 room 4
load 0x3ffe8914, len 8, room 12 
tail 8
chksum 0x42
csum 0x42
����n��{��o|��d�p{l��{�d�l�
                               �d`�

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 3328c66f2c1c4b83f96b3c869b2167bb2ab183a2
	SSL: false
	modules: adc,bit,file,gpio,http,i2c,net,node,rtctime,sntp,spi,tmr,u8g,uart,wifi
 build 	built on: 2016-08-28 14:34
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
>
Starting
bootreason: 2, 2
```
### Test code
```Lua
print("Starting")
local boot_reason, reset_cause = node.bootreason()
local wakeup = false
print("bootreason: " .. boot_reason .. ", " .. reset_cause)
if reset_cause == 5 or reset_cause == 6 then wakeup = true end
if wakeup then print("Waking up from sleep") end
--
if not wakeup then
    wifi.setmode(wifi.STATION)
    wifi.sta.config("X","Y")
    sntp.sync("tak.cesnet.cz")
else
    print("Calling wifi.setmode(wifi.NULLMODE)")
    wifi.setmode(wifi.NULLMODE)
end
```

and then type in the Lua console:
```
rtctime.dsleep(1000000, 4)
```

### NodeMCU version
branch: dev
	commit: 3328c66f2c1c4b83f96b3c869b2167bb2ab183a2

### Hardware
plain ESP8266 with GPIO16 wired to RST

dnc40085, 2016-08-29:
Can you please try calling `wifi.nullmodesleep(false)` before `wifi.setmode(wifi.NULLMODE)`.
I mistakenly didn't test SDK function `wifi_fpm_auto_sleep_in_null_mode()` with deep sleep.

joysfera, 2016-08-29:
Before I try that I'd like to mention that there seems to be some sort of timing issue because when I call the ```wifi.setmode(wifi.NULLMODE)``` manually few seconds later afer waking up it seems to work fine.

And a real work-around for this issue is to simply disable the WiFi modem before falling to dsleep - that way I don't have to set it after waking up.

closed 2019-06-21 by stale[bot]