nodemcu/nodemcu-firmware #1482: Native WebSocket Server

luismfonseca, 2016-08-29:
### Missing feature
A WebSocket Server as part of the `websocket` module.

### Justification
A native implementation uses as little heap as possible and is considerably faster.

The server would allow browsers to connect to the NodeMCU as it would function as a server.

### Proposed implementation

This would be an extension of the existing [websocket](http://nodemcu.readthedocs.io/en/dev/en/modules/websocket/) module.

Syntax could be as follows (@creationix perhaps you could be kind enough to provide me with your input):

```lua
wss = websocket.createServer(80,
  function(ws, path) -- ws is the websocket.client already implemented
    print('new client connected!')
    ws:on("close", function() print('got ws close') end)
    ws:on("receive", function(_, msg) print(msg) end)
  end
)

wss:close()
```

Also, my current kinda-works implementation continues using the deprecated `espconn` layer as I'm still waiting for progress in #1379 and #1435.

Should I make a MR or wait for those?

### Workarounds
There is one implementation in Lua for the NodeMCU [[1]](https://github.com/creationix/nodemcu-webide).

djphoenix, 2016-08-29:
About espconn - I propose to do server-side work with plain lwip because espconn not have secure server anyway. For clients espconn provides one way to get SSL connection. For servers you can only listen TCP.

creationix, 2016-08-29:
@luismfonseca looks good as far as I can tell.  One suggestion might be to add support for subprotocols (part of the handshake), but considering the limited nature of these boards, it's probably not needed and is better to keep things simple.

I haven't had time to look closely as the client API, but I do know that I'll need a way to tell apart text and binary frames in both sending and receiving.

luismfonseca, 2016-08-30:
@djphoenix: Ok, I can start by moving everything to use plain lwIP.

But the mbedTLS is only present in the #1435, so a secure client and server are blocked by that. You could split the merge request to add mbedTLS separately.

@creationix Thanks. I'm also inclined to leave subprotocols to future work as it seems kinda overkill.

The client API does have a way to difference binary and text messages (although it exposes the protocol specification). You could do a `ws:send(data, opcode)` and have `opcode = 1` for text (implicit default) or `opcode = 2` for binary.

creationix, 2016-08-30:
@luismfonseca looks good.

marcelstoer, 2018-02-24:
No progress in a long time -> closing

closed 2018-02-24 by marcelstoer