nodemcu/nodemcu-firmware #1483: Newly compiled firmware only spits out garbage through UART

ameeuw, 2016-09-02:
### Expected behavior
Booting up ad replying with NodeMCU version

### Actual behavior
Constantly puts out garbage through UART

### NodeMCU version
master branch (last working commit is b580bfe79e6e73020c2bd7cd92a6afe01a8bc867)

### Hardware
ESP 12E Witty Board

dnc40085, 2016-09-02:
The new sdk, version 1.5.4.1, requires `esp_init_data_default.bin` to be burned as well.
You can find instructions [here](http://nodemcu.readthedocs.io/en/latest/en/flash/) on how and where to flash the binary.

marcelstoer, 2016-09-02:
See #1010 and http://nodemcu.readthedocs.io/en/latest/en/support/

closed 2016-09-02 by marcelstoer

dnc40085, 2016-09-02:
@marcelstoer I am aware of the policy for updating the master branch, but I think just this once we should make an exception to this rule and update the issue template to include a message about `esp_init_data_default.bin` and the need to flash it along side the other bins.

dnc40085, 2016-09-02:
@marcelstoer I had another idea... would it be possible to include `esp_init_data_default.bin` in the build from https://nodemcu-build.com/?
Maybe it would prevent people who use the cloud build service from getting this error.

marcelstoer, 2016-09-02:
\<OT\>
The idea is good BUT:
- I'd have to add an option to select/define flash size because of the [memory address for the init data block](http://nodemcu.readthedocs.io/en/latest/en/flash/#upgrading-firmware).
- You couldn't build a firmware and flash it to multiple modules with different flash sizes.
- The binary wouldn't "match" with what is documented every where around the Internet regarding how to flash.
- People would poke me directly even more frequently when something fails even though [my FAQ](https://nodemcu-build.com/faq.php#get-help) is quite clear IMO. 

\</OT\>