nodemcu/nodemcu-firmware #1484: Support TLS SNI or handle error gracefully

marcelstoer, 2016-09-04:
### Expected behavior
Either support [TLS SNI](https://en.wikipedia.org/wiki/Server_Name_Indication) or at least handle errors gracefully.

Someone wanted to provide a fix for axTLS at https://github.com/dsheets/axtls/issues/2 but it's still open. Can our firmware at least handle the error?

### Actual behavior
Crash.

> client handshake start.
> client handshake ok!
> client's data invalid protocol
> Error: invalid protocol message
> Fatal exception 28(LoadProhibitedCause):
> epc1=0x4000df1b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x01000000, depc=0x00000000
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)

### Test code
```Lua
http.get("https://nodemcu-build.com/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```
Other than its use of SNI the certificate for the domain is fine: https://www.ssllabs.com/ssltest/analyze.html?d=nodemcu-build.com

### NodeMCU version
Latest `dev`.

### Hardware
NodeMCU devkit V2/1.0.

jmattsson, 2016-09-05:
I'm getting the definite impression we need to get some more focus on the shift to mbedTLS. I'd do it if I had time for it, but at the moment I do not.

marcelstoer, 2016-09-18:
I'll close this. https://github.com/nodemcu/nodemcu-firmware/issues/1485#issuecomment-246176969 showed that the cause is unrelated to SNI. https://nodemcu-build.com/jumbotron-narrow.css works just fine.

closed 2016-09-18 by marcelstoer