nodemcu/nodemcu-firmware #1485: HTTP to https://google.com fails

marcelstoer, 2016-09-04:
### Expected behavior
Print google.com HTML markup.

### Actual behavior
Request failure.

> client handshake start.
> client handshake ok!
> client handshake start.
> client handshake failed
> Error: invalid protocol message
> HTTP request failed

The Google SSL certificate supports TLS 1.0-1.2 and two of the four cipher suites our axTLS lib supports are supported by the Google cert.

### Test code
Provide a minimal and isolated test which will reproduce the problem.
```Lua
http.get("https://google.com", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```
The HTTP 301 from google.com to www.google.com works fine behind the scenes. It appears that the handshake for google.com is ok but for www.google.com not ("client handshake ok!" and then "client handshake failed").

### NodeMCU version
Latest `dev`.

### Hardware
NodeMCU devkit v2/1.0.

WilliamOckham, 2016-09-07:
Is there a working example of NodeMCU connecting to a web server that requires server verification? I haven't been able to get it to work at all using either the http module or crafting the request myself with the tcp client.

mikewen, 2016-09-07:
I think the problem is the redirect or CDN. Try some other plain servers.

http module works well for my WiReboot <http://www.WiReboot.com>project.
I use NodeMCU to POST to my web App hosted on Google App Engine:
https://wirebootapp.appspot.com/

marcelstoer, 2016-09-07:
@WilliamOckham Correct me if I'm wrong but AFAICT your comment has nothing to do with this issue, see http://nodemcu.readthedocs.io/en/latest/en/support/.

> Try some other plain servers.

No I won't 😉 Whatever the cause this is specifically about google.com.

luismfonseca, 2016-09-07:
I don't think it's the redirect (try: `google.co.uk` which doesn't redirect).

This issue also manifests itself in the `master`.

Also, it's not just `google.com`. `microsoft.com` also fails and others, so it indicates a problem with our certificate validation...

mikewen, 2016-09-07:
Second thought, right. The problem could be SSL for naked domain , or could
be wildcat SSL.

WilliamOckham, 2016-09-08:
I could be wrong, but I believe that the behavior is not limited to google.com and that the redirect happens before the server verification process completes. I've seen similar behavior connecting to Amazon.com and several other sites. I can't tell for sure that the failures are caused by the same issue. Thus, I would like to gather network traces for a successful server verification process for comparison. Unfortunately, the only TLS-enabled sites that I have successfully connected to have been sites that don't do server verification. 

All of my tests have been performed with a known good cert installed on the device and net.cert.verify returns true before every request. I am not asking how to connect to an https enabled site. I am trying to help diagnose what looks to be a serious bug that is blocking a current project I am working on.

jmattsson, 2016-09-08:
Marcel, can you try a NodeMCU with [mbedTLS](https://github.com/DiUS/nodemcu-firmware/tree/esp_mbedtls)? I'm having internal DNS issues that seems to be messing up my testing :/

marcelstoer, 2016-09-10:
No Johny, doesn't work either.

> client handshake start.
> client handshake ok!
> client's data invalid protocol
> Reason:[-0x7880]
> client handshake start.
> client handshake failed!
> Reason:[-0x7200]
> HTTP request failed

However, in that branch there seem to be other issues as well.

```
http.get("https://nodemcu-build.com/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```
Resets the device

> client handshake start.
> client handshake ok!
> client's data invalid protocol
> Reason:[-0x7200]
> Fatal exception 28(LoadProhibitedCause)

Even non-SSL requests fail

```
http.get("http://frightanic.com/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```

> HTTP request failed

jmattsson, 2016-09-11:
Hmm, I must've screwed something up! What are you using to generate these "client handshake start." etc messages?

marcelstoer, 2016-09-11:
Those are regular debug messages.

luismfonseca, 2016-09-11:
> However, in that branch there seem to be other issues as well.

Have you tested this on master? It seems that it also resets the device on `http://frightanic.com`... :/

marcelstoer, 2016-09-11:
I had a hunch and suspected that the `code < 0` cases (i.e. "HTTP request" failed in my MVCE) were related to response size. In my previous test run we already established that this issue seems unrelated to SSL/TLS. 

I was right. A few tests:

- http://frightanic.com/jquery-highlight.js ✅ , 2.7 KB
- http://frightanic.com 🔴 , 17,7 KB
- http**s**://nodemcu-build.com 🔴 ♻️ , 6.6 KB
- http**s**://nodemcu-build.com/about.php 🔴 ♻️ , 2.8 KB
- http**s**://nodemcu-build.com/jumbotron-narrow.css ✅ , 433 B
- http**s**://www.google.com/, 🔴 , 66.9 KB
- http**s**://clients5.google.com/pagead/drt/dn/, ✅ , 270 B

🔴 : fails
♻️ : reset

The debug messages stating that "client handshake failed" may be misleading.

jmattsson, 2016-09-14:
I played a bit more with the mbedtls branch (and fixed one issue), but there are more gremlins in there. Something eats data between the lwip layer and the espconn-callback. As in, if I printf in `lwIPSocket.c:recv_tcp()` I see SSL packets coming in. None of the data gets delived to the espconn secure receive callback though, even when the SSL handshake completes.

I freely admit I don't understand how they're using the lwIP socket API and where they would've put the stack for the second "thread" that afaiu the socket API needs...

TL;DR: mbedtls on non-RTOS considered suspect at this point

djphoenix, 2016-09-14:
@jmattsson I really think mbedTLS itself is not suspect. I used it on plain NONOS SDK, and it works as well, including handshakes and long data exchange.

I will play with mbedTLS (and espconn) in nodemcu right after finish work at #1379 (maybe today).

marcelstoer, 2016-10-20:
This issue took a number of twists and turns with SSL/non-SSL issue hints I'd like to close it in favor of #1546.

closed 2016-10-20 by marcelstoer