nodemcu/nodemcu-firmware #1490: Doc link for master points to dev.

devsaurus, 2016-09-06:
### Expected behavior 
Link in README.md points to docs consistent with master branch.

### Actual behavior
Link points to latest which shows dev-related content atm. Eg. the new sdcard chapter.
I thought up to now that latest is an alias for master, but apparently not.

### NodeMCU version
Current repo.

marcelstoer, 2016-09-06:
`latest` is an alias for `dev`. The default, however, is `master`.
- http://nodemcu.readthedocs.io/ -> master
- http://nodemcu.readthedocs.io/en/latest/ -> dev
- the rest is obvious
Invalid or misleading internal links such as "Edit on GitHub" are bugs with either MkDocs or Read the Docs.

devsaurus, 2016-09-07:
> Alternatives?

Would an intended & static difference in `README.md` on the two branches be enough? Or does this generate merge conflicts?

marcelstoer, 2016-09-07:
I could live with that if there actually _were_ merge conflicts 😉 It would remind us at every master drop that we need to keep the URLs different. What I'm worried about is a silent merge/replacement. However, it'd be worth a try. Can't be worse than the status quo.

commit 8e48483c (2016-09-09): Change RTD links to point to `master`

Fixes #1490
closed 2016-09-09 by marcelstoer