nodemcu/nodemcu-firmware #1494: CJSON decode limited to 250 characters

Jonathan411, 2016-09-07:
A json string with 250 characters or less decodes correctly, but if the string of json data is 251 characters or more in length then an error is raised:  LUA interpreter error detected!

The following code works:

```
jstr='{"aa":"1","bb":"2","cc":"3","dd":"4","ee":"5","ff":"6","gg":"7","hh":"8","ii":"9","jj":"10","kk":"11","ll":"12","mm":"13","nn":"14","oo":"15","pp":"16","qq":"17","rr":"18","ss":"19","tt":"20","uu":"21","vv":"22","w":"23","x":"24","y":"25","z":"26"}'
for k,v in pairs(cjson.decode(jstr)) do print(k,v) end

```

The following code generates an error (one additional "w" character):

```
jstr='{"aa":"1","bb":"2","cc":"3","dd":"4","ee":"5","ff":"6","gg":"7","hh":"8","ii":"9","jj":"10","kk":"11","ll":"12","mm":"13","nn":"14","oo":"15","pp":"16","qq":"17","rr":"18","ss":"19","tt":"20","uu":"21","vv":"22","ww":"23","x":"24","y":"25","z":"26"}'
for k,v in pairs(cjson.decode(jstr)) do print(k,v) end

```

Is the 250 character limit a feature or a bug? I couldn't find any documentation which mentions a limit on the size of the json input string.

luismfonseca, 2016-09-07:
This doesn't seem to be a limitation of the `cjson` module, but rather the max length that Lua String can be declared (without concatenation).

The following code works:
```lua
jstr = '"aa":"1","bb":"2","cc":"3","dd":"4","ee":"5","ff":"6","gg":"7","hh":"8","ii":"9","jj":"10","kk":"11","ll":"12","mm":"13","nn":"14","oo":"15","pp":"16","qq":"17","rr":"18","ss":"19","tt":"20","uu":"21","vv":"22","w":"23","x":"24","y":"25","z":"26"'

jstr = '{'..jstr..',"xyz":"1231231231231"}'

print(string.len(jstr) .. '\n') -- 270

for k,v in pairs(cjson.decode(jstr)) do print(k,v) end
```

Jonathan411, 2016-09-08:
Not a NodeMCU issue then.  Thanks.

closed 2016-09-08 by Jonathan411

luismfonseca, 2016-09-08:
Well, I think it sorta is, in the sense that the `250` is the limit configured in NodeMCU...

If you run Lua on a computer, it doesn't have this limitation. Well, at least not `250`. So I suppose that this was mostly a decision based on memory constraints.