nodemcu/nodemcu-firmware #1495: Dump the stack on exception in DEVELOP_VERSION

pjsg, 2016-09-08:
### Missing feature
When DEVELOP_VERSION is enabled, the firmware should dump out the contents of the stack on an exception (rather than just restarting the platform).
### Justification
This will make debugging much easier. The ESP8266 version of the Arduino code has this capability (https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_postmortem.c) but it looks as though it needs lots of other components as well.....

### Workarounds
Nope. I'm beating my head against a brick wall. It is nice when I stop.

jmattsson, 2016-09-08:
Looks like it's relying on some of the gdb_stub stuff?

Also, the stack frame decoding looks rather primitive, but I guess that's a lot better than what we currently have!

pjsg, 2016-09-09:
Maybe I should just integrate https://github.com/espressif/esp-gdbstub first (obviously under a config define). Thoughts? Or has someone else already done it?

pjsg, 2016-09-20:
See #1505 to integrate the gdbstub....

closed 2019-06-21 by stale[bot]