nodemcu/nodemcu-firmware #1501: libc is missing functions (c_snprintf and vsnprinf)

dnc40085, 2016-09-16:
### Missing feature
The functions `c_snprintf` and `vsnprintf` are missing from `app/libc/stdio.c`.

### Justification
Using `c_snprintf` rather than `c_sprintf` would be safer since maximum buffer length can be specified, preventing possibility of buffer overflow.

### Workarounds
The SDK has two undocumented functions (`ets_snprintf` and `ets_vsnprintf`) available, but `ets_vsnprintf` is lacking some [format specifiers](http://www.cplusplus.com/reference/cstdio/printf/) ( fF eE gG aA) making it incompatible with floating point values.

devsaurus, 2016-09-17:
> `ets_vsnprintf` is lacking some format specifiers ( fF eE gG aA) that make it incompatible with floating point builds

Seems that this was the reason for replacing `ets_sprintf` by a local implementation in float builds: https://github.com/nodemcu/nodemcu-firmware/blob/master/app/libc/c_stdio.h#L59.

`c_snprintf` would need the same logic, I assume.

jmattsson, 2016-09-17:
With RTOS SDK and the IDF we get to use actual libc functions. It's grand. :)

pjsg, 2017-10-29:
I notice that the %o specifier doesn't work in the current dev branch (it seems to format as decimal). If you look at the examples above, you will see

```
    Octal(%o):2329
```

This seems an unlikely output. I *think* that this comes from the underlying os_sprintf implementation.  It probably ought to be reported to espressif

devsaurus, 2019-07-21:
I expect that `snprintf` / `vsnprintf` will be added with #2838.

This issue should be revisited, e.g. in the light of https://github.com/nodemcu/nodemcu-firmware/blob/9f8b74debd9f030d63f8b4db3db07bc15b748765/app/modules/wifi.c#L3

closed 2020-07-30 by stale[bot]