nodemcu/nodemcu-firmware #1502: Several issues with spiffsimg

marcelstoer, 2016-09-18:
Triggered by a question from one of our users on Stack Overflow set out to use `spiffsimg` for the first time. I ran into 3 issues which may all be rooted in a lack of documentation - even though https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/spiffs.md is quite extensive.

**Which `spiffsimg`**
It all started with the question from where to get `spiffsimg`. The docs don't say. I knew about [`/tools/spiffsimg`](https://github.com/nodemcu/nodemcu-firmware/tree/master/tools/spiffsimg) but I also knew about https://github.com/DiUS/spiffsimg.
I went with ours because it's newer.

**How to build**
So, I wanted to build it from `/tools/spiffsimg`.
```
$ make
cc -g -Wall -Wextra -Wno-unused-parameter -Wno-unused-function -I. -I../../app/spiffs -DNODEMCU_SPIFFS_NO_INCLUDE --include spiffs_typedefs.h main.c ../../app/spiffs/spiffs_cache.c ../../app/spiffs/spiffs_check.c ../../app/spiffs/spiffs_gc.c ../../app/spiffs/spiffs_hydrogen.c ../../app/spiffs/spiffs_nucleus.c  -o spiffsimg
In file included from main.c:44:
In file included from ../../app/spiffs/spiffs.h:14:
../../app/spiffs/spiffs_config.h:17:10: fatal error: 'user_config.h' file not found
#include "user_config.h"
```
I fixed the include path and finally got a `spiffsimg` (plus a `spiffsimg.dSYM` folder).

**Invalid image file name**
The docs say

> `-f` specifies the filename for the disk image. '%x' will be replaced by the calculated offset of the file system.

Yet, the following command produced a `0x-msr.img` in the current folder (`/tools/spiffsimg`):

`$ ./spiffsimg -f 0x-msr.img -c 262144`
 
 

_-> @jmattsson @pjsg Which one is a user error, which one a bug in the code and which one a bug in the docs?_

pjsg, 2016-09-18:
Some of this is due to an issue that I had forgotten about -- the spiffsimg tool needs a C compiler inside the docker container that builds for the host platform. In order not to break things, the main makefile does not build the spiffsimg tool automatically (unless there is a local directory which contains the file to be put into an image). 

The fix for that is to always build spiffsimg (now that gcc is in the stock docker container). I'll do a PR today.

I suspect that the documentation does not distinguish between the case where the image is built automatically (if the local/fs directory exists) and when you invoke it manually. I'll clean that up too.

As to the substitution of the %x -- this should only work if the size of the flash used by the image is passed. That requires two fixes -- one to the docs to explain this, and one to the code to throw an error in this case.

On the plus side, you actually tried it!

pjsg, 2016-09-18:
On further investigation, @devsaurus updated the spiffs_config.h file to include user_config.h. Since the spiffimg tool was not being built by default, he couldn't know that that would break it.

marcelstoer, 2016-09-18:
Thanks Philip for your feedback.

> spiffsimg tool needs a C compiler inside the docker container that builds for the host platform

I know but I wasn't using that with Docker. I executed `make` in the OS X bash but I was in `/tools/spiffsimg` rather than in `/tools`.

> As to the substitution of the %x -- this should only work if the size of the flash used by the image is passed.

Yep, I wondered where the magic in the calculation came from 😉 

> Since the spiffimg tool was not being built by default, he couldn't know that that would break it.

It's indeed helpful to always build it, thanks.

pjsg, 2016-09-18:
Ah -- let me see if my change allows you to build spiffsimg natively (outside docker).

marcelstoer, 2016-09-21:
I wasn't fast enough with testing before Johny merged #1503, sorry 😞 Of course, that PR was a step forward, though. Thanks.

I still see two issues:

**Building from `/tools` fails**
I did this on OS X. So, no real surprise but could the script catch and handle this?
```
~/Data/NodeMCU/nodemcu-firmware/tools (dev) > make
if [ -r ../bin/spiffs-4mb.dat ]; then rm -f ../bin/$(cat ../bin/spiffs-4mb.dat)-4mb.bin; fi;   if [ -r ../bin/spiffs-32mb.dat ]; then rm -f ../bin/$(cat ../bin/spiffs-32mb.dat)-32mb.bin; fi;   if [ -r ../bin/spiffs-8mb.dat ]; then rm -f ../bin/$(cat ../bin/spiffs-8mb.dat)-8mb.bin; fi; 
rm -f ../bin/spiffs*.dat
gcc -g -Wall -Wextra -Wno-unused-parameter -Wno-unused-function -I. -I../../app/spiffs -I../../app/include -DNODEMCU_SPIFFS_NO_INCLUDE --include spiffs_typedefs.h main.c ../../app/spiffs/spiffs_cache.c ../../app/spiffs/spiffs_check.c ../../app/spiffs/spiffs_gc.c ../../app/spiffs/spiffs_hydrogen.c ../../app/spiffs/spiffs_nucleus.c  -o spiffsimg
rm -f ./spiffsimg/spiffs.lst
echo "" >> ./spiffsimg/spiffs.lst
/bin/sh: xtensa-lx106-elf-objdump: command not found
^C
```

`spiffsimg` was correctly built. I then deleted it and did the same from the `spiffsimg` folder.

```
~/Data/NodeMCU/nodemcu-firmware/tools/spiffsimg (dev) > make
cc -g -Wall -Wextra -Wno-unused-parameter -Wno-unused-function -I. -I../../app/spiffs -I../../app/include -DNODEMCU_SPIFFS_NO_INCLUDE --include spiffs_typedefs.h main.c ../../app/spiffs/spiffs_cache.c ../../app/spiffs/spiffs_check.c ../../app/spiffs/spiffs_gc.c ../../app/spiffs/spiffs_hydrogen.c ../../app/spiffs/spiffs_nucleus.c  -o spiffsimg
~/Data/NodeMCU/nodemcu-firmware/tools/spiffsimg (dev) >
```

All good.

**Tutorial for the masses**
IMO `spiffs.md` should have a recipe-style section (step-by-step bullet list) for our largest user base - those using the cloud builder. I don't think it's user-friendly to have them wade through the 'Technical Details' section to figure out how to build and use `spiffsimg`. IMO it should answer a simple question: "I just downloaded a NodeMCU binary, what do I have to do to get a SPIFFS image for flashing?". Even better would be if we had _two_ recipes, one for the build-manually-or-with-Docker crowd and one for the others. 

Sounds feasible?

PR #1503 (2016-09-21): Fixes the spiffsimg issues (#1502)
Fixes #1502 

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This fixes the bugs reported in #1502. In particular, now that the stadard docker container includes the
local `gcc`, the `spiffsimg` tool is always built. This will prevent any future breakage of the tool (as it will fail the build). `spiffsimg` now fails if you t [... cut]
closed 2019-06-21 by stale[bot]