nodemcu/nodemcu-firmware #1507: myspiffs_size

MitchBradley, 2016-09-20:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

spiffs.c: myspiffs_size() should return the file size but ...

### Actual behavior

... it always returns 2.

### Test code

Lua apparently does not call myspiffs_size() but I have a custom build that exercises it.

The problem is obvious; the final two arguments to SPIFFS_lseek() are backwards - _whence, off_ when they should be _off, whence._  Here is the patch to fix it:

```

diff --git a/app/spiffs/spiffs.c b/app/spiffs/spiffs.c
index d481d6e..d11c39c 100644
--- a/app/spiffs/spiffs.c
+++ b/app/spiffs/spiffs.c
@@ -281,8 +281,8 @@ int myspiffs_rename( const char *old, const char *newname ){
 }
 size_t myspiffs_size( int fd ){
   int32_t curpos = SPIFFS_tell(&fs, (spiffs_file) fd);
-  int32_t size = SPIFFS_lseek(&fs, (spiffs_file) fd, SPIFFS_SEEK_END, 0);
-  (void) SPIFFS_lseek(&fs, (spiffs_file) fd, SPIFFS_SEEK_SET, curpos);
+  int32_t size = SPIFFS_lseek(&fs, (spiffs_file) fd, 0, SPIFFS_SEEK_END);
+  (void) SPIFFS_lseek(&fs, (spiffs_file) fd, curpos, SPIFFS_SEEK_SET);
   return size;
 }
 #if 0

```

### NodeMCU version

master  84c005e85

### Hardware

It's hardware-independent

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-09-20:
Thanks for highlighting this bug and for the proposed correction.

The overall situation changed quite a bit on `dev` branch since the API to `myspiffs_` functions got replaced by the vfs layer that came with #1397: Code like yours will face the challenge to adopt to vfs API, latest when `dev` gets merged into `master` with the [upcoming drop](https://github.com/nodemcu/nodemcu-firmware/milestone/2) beginning of October.
The equivalent function is [`vfs_size()`](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/platform/vfs.h#L83), but it's currently not implemented [for SPIFFS](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/spiffs/spiffs.c#L298) (just because it wasn't required until now). Adding a flavour for SPIFFS should be no big deal. This time in the correct way, as you pointed out :wink:

I'll prepare a PR soon.

jmattsson, 2016-09-21:
@devsaurus would you mind checking the uses of seek in the file module as well? I have a vague recollection of having seen oddity in this area last week, but I got sidetracked before I had a chance to investigate it. Seeing this bug report jogged my mind a bit.

devsaurus, 2016-09-23:
> would you mind checking the uses of seek in the file module as well?

Will do.

devsaurus, 2016-09-28:
@jmattsson no obvious issues with `file.seek()` observed:

```lua
-- 01
-- 02
-- 03
-- 04
-- 05
-- 06
-- 07
-- 08
-- 09
-- 10
-- 11
-- 12
-- 13
-- 14
-- 15

file.open( "file_seek.lua" )
for line = 1,6 do
   file.seek( "set", (line-1) * 6 + 3 )
   num = file.read( 2 )
   if tonumber( num ) ~= line then error( "seek_set: wrong content in line "..line..": "..num ) end
end

file.seek( "set", (10-1) * 6 + 3 )
for line = 10,15 do
   num = file.read( 2 )
   if tonumber( num ) ~= line then error( "seek_cur: wrong content in line "..line..": "..num ) end
   file.seek( "cur", 4 )
end

for line = 91,96 do
   file.seek( "end", -((line-91) * 6 + 3) )
   num = file.read( 2 )
   if tonumber( num ) ~= line then error( "seek_end: wrong content in line "..line..": "..num ) end
end

file.close()

-- 98
-- 97
-- 96
-- 95
-- 94
-- 93
-- 92
-- 91
```

jmattsson, 2016-09-29:
@devsaurus Thanks, I must've had something else being wrong. I tried your test cases on my ESP32 and they worked there too.

devsaurus, 2016-10-03:
Missing functionality has been re-added on `dev` branch.

PR #1516 (2016-10-03): implement file.size for spiffs
Fixes #1507.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

This PR re-adds querying the size of an opened file to SPIFFS.

Committers supporting this PR:
closed 2016-10-03 by devsaurus