nodemcu/nodemcu-firmware #1508: Enduser Setup is not picking up custom html file

pmpbar, 2016-09-20:
### Expected behavior
Enduser setup is not picking up "enduser_setup.html" in fs.

### Actual behavior
Fall back file is being loaded.

### Test code
```Lua
  wifi.setmode(wifi.STATIONAP);
  local board_ssid = "board_" .. node.chipid();
  wifi.ap.config({ssid=board_ssid, auth=wifi.OPEN});
  wifi.ap.setip({ip="192.168.1.1",
    netmask="255.255.255.0",
    gateway="192.168.1.1"});
  enduser_setup.manual(true);
  enduser_setup.start(
  function()
      print('\n====================================')
      print('ESP8266 mode is: ' .. wifi.getmode())
      print('MAC address is: ' .. wifi.ap.getmac())
      print('IP is '.. wifi.sta.getip())
      print('====================================')
      app.start()
  end,
  function(err, str)
      print("enduser_setup: Err #" .. err .. ": " .. str);
  end
  );

```
### NodeMCU version
Master branch, latest release from cloud build with modules:  [adc, cjson, crypto, encoder, enduser_setup, file, gpio, http, i2c, mdns, mqtt, net, node, pwm, spi, tmr, uart, wifi].

### Hardware
ESP-12E, 4MB

jmattsson, 2016-09-21:
Tagging in @robertfoss ; I'm currently busy over on the ESP32 side...

marcelstoer, 2016-09-22:
The sample code doesn't show that the file actually exists under the expected name at the expected location. Please include

```lua
if file.exists("enduser_setup.html") then
    print("Detected custome page for EUS")
end
```
and show the output.

sergbod, 2016-09-28:
For me also doesn't work. Costoized page is detected, but fall back file is loaded.
Also observed very-very slow loadin of default enduser setup page if not manual configuration selected. tested on differetn browsers, like Chrome and IE and for desctop and mobile.
HW: ESP-07 [1MB flash]
Master branch, latest release. Used 10 modules+SSL+DEBUG: bit, enduser_setup, file, http, i2c, net, node, tmr, uart, wifi.

robertfoss, 2016-10-05:
Hmm. This should be investigated.

I don't have enough free cycles to take this up now though.

HeribertB, 2016-10-28:
Appears to me it's "err". vfs_lseek does return the position, not an error code, so the clause 

           if (!f || err != VFS_RES_OK || err2 != VFS_RES_OK)

yields FALSE anyway.

devsaurus, 2016-10-29:
Good catch, @HeribertB! It's my bad that the conditional logic got broken with #1397. I'll submit a PR soon for this particular issue.
Although the original report for `master` on 20-Sep should not be affected by this. #1397 was merged to `master` with the [Post-1.5.4.1 drop](https://github.com/nodemcu/nodemcu-firmware/milestone/2) on 01-Oct.

closed 2016-11-20 by marcelstoer