nodemcu/nodemcu-firmware #1513: Simple RGB/GRB mixup in WS2812 module

georeb, 2016-09-26:
The WS2812 module does not follow the recognised RED GREEN BLUE (RGB) sequence, but is instead GREEN RED BLUE (GRB) for some reason. This is causing unnecessary problems requiring extra code to patch the mixup. (Unless there is a reason for this??)

Below shows the write command that is at fault...
```Lua
ws2812.write(string.char(g, r, b))
```
### NodeMCU version
Master

### Hardware
ESP8266-12E & WS2812 LED

marcelstoer, 2016-09-26:
I'm sure this isn't an arbitrary glitch but you'd have to discuss with @dereulenspiegel and @Alkorin  (originator and maintainer). [The docs are clear](https://nodemcu.readthedocs.io/en/master/en/modules/ws2812/#ws2812write):

> Send data to one or two led strip using its native format which is generally Green,Red,Blue for RGB strips and Green,Red,Blue,White for RGBW strips.

Alkorin, 2016-09-26:
Yes, we support N-bytes leds strips. They could be RGB, GRB, GRBW, or even 'Warm white, Pure white, Amber' like SK6812WWA. The library has no knowledge about the real color map, it just handle a bunch of bytes. You could add your own wrapper to mixup color for your own needs (which depends on your stripe). 

Personally, I handle hundreds of leds per ESP8266 and I don't really want to first mixup my colors to the RGBW pattern and then let the library to translate it to GRBW. I write my algorithms to output GRBW colors directly ;)

georeb, 2016-09-26:
Okay, understood. I didn't realise there was such variation! Having worked with many image processing programs, which use RGB as standard, I (wrongly) assumed that it was an oversight in the code, thanks for clearing this up.

closed 2016-09-26 by georeb